lvgl/lv_binding_micropython #89: Error using gen_mpy.py on macOS

adahl5, 2020-07-26:
Hello, I'm getting an error when I try to use gen_mpy.py on macOS, the error I'm getting is a ParseError from the pycparser. I'm getting the error `pycparser.plyparser.ParseError: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/11.0.3/include/stdarg.h:14:27: before: va_list` I'm in over my head and could use some help. Thanks.

adahl5, 2020-07-26:
I think the issue was not including the pycparser fake_libc_include. But now I'm running into an issue where g++ is generating a lot of errors such as `typedef redefinition with different types ('int' vs 'unsigned int')`

amirgon, 2020-07-26:
Hi @adahl5 !

> I'm getting an error when I try to use gen_mpy.py on macOS, the error I'm getting is a ParseError from the pycparser. I'm getting the error `pycparser.plyparser.ParseError: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/11.0.3/include/stdarg.h`

I'm not familiar with macOS, but I can see you use clang. Could you try gcc?  
pycparser is not supposed to process the OS specific standard libraries. They are replaced by fake_libc_include. So I wouldn't expect to see `clang/11.0.3/include/stdarg.h` error from pycparser. Need to see why it's included instead of fake_libc_include.

> I think the issue was not including the pycparser fake_libc_include. But now I'm running into an issue where g++ is generating a lot of errors such as `typedef redefinition with different types ('int' vs 'unsigned int')`

You are supposed to include fake_libc_include only for pycpareser preprocessing.  
When building the C code you should use the regular libraries.

> But now I'm running into an error with FreeRTOS.h where most the #endif's generate errors like `FreeRTOS.h:916:3: error: #endif without #if`

What was the full command line?

> Now receiving the error `pycparser.plyparser.ParseError: ~/esp/esp-idf/components/esp_common/include/esp_err.h:80:118: before: __attribute__` I'm using xtensa-esp32-elf-cpp for preprocessing

pycparser does not know how to handle `__attribute__`. 
For that purpose espidf module [defines it explicitly for pycparser](https://github.com/lvgl/lv_binding_micropython/blob/30b9445a28be2bbeada39183aac138e936341d4b/driver/esp32/espidf.h#L12).

**Please provide more details about what you are doing, what changes you did in the code / makefiles, the full command line that failed etc.**

adahl5, 2020-07-27:
Here is the full command line.
`make PYCPARCER=1
find: /Users/adam/esp/esp-idf/components/xtensa-debug-module/include: No such file or directory
find: /Users/adam/esp/esp-idf/components/smartconfig_ack/include: No such file or directory
ESPIDFMOD-GEN build/espidfmod/test.c
mkdir -p build/espidfmod/
/Users/adam/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-cpp -DPYCPARSER -nostdinc -w -Wno-deprecated-declarations  -I/Users/adam/esp/esp-idf/components/bootloader_support/include -I/Users/adam/esp/esp-idf/components/bootloader_support/include_bootloader -I/Users/adam/esp/esp-idf/components/console -I/Users/adam/esp/esp-idf/components/driver/include -I/Users/adam/esp/esp-idf/components/driver/include/driver -I/Users/adam/esp/esp-idf/components/efuse/include -I/Users/adam/esp/esp-idf/components/efuse/esp32/include -I/Users/adam/esp/esp-idf/components/esp32/include -I/Users/adam/esp/esp-idf/components/espcoredump/include -I/Users/adam/esp/esp-idf/components/soc/include -I/Users/adam/esp/esp-idf/components/soc/esp32/include -I/Users/adam/esp/esp-idf/components/heap/include -I/Users/adam/esp/esp-idf/components/log/include -I/Users/adam/esp/esp-idf/components/nvs_flash/include -I/Users/adam/esp/esp-idf/components/freertos/include -I/Users/adam/esp/esp-idf/components/esp_ringbuf/include -I/Users/adam/esp/esp-idf/components/esp_event/include -I/Users/adam/esp/esp-idf/components/tcpip_adapter/include -I/Users/adam/esp/esp-idf/components/lwip/lwip/src/include -I/Users/adam/esp/esp-idf/components/lwip/port/esp32/include -I/Users/adam/esp/esp-idf/components/lwip/include/apps -I/Users/adam/esp/esp-idf/components/lwip/include/apps/sntp -I/Users/adam/esp/esp-idf/components/mbedtls/mbedtls/include -I/Users/adam/esp/esp-idf/components/mbedtls/port/include -I/Users/adam/esp/esp-idf/components/mdns/include -I/Users/adam/esp/esp-idf/components/mdns/private_include -I/Users/adam/esp/esp-idf/components/spi_flash/include -I/Users/adam/esp/esp-idf/components/ulp/include -I/Users/adam/esp/esp-idf/components/vfs/include -I/Users/adam/esp/esp-idf/components/xtensa-debug-module/include -I/Users/adam/esp/esp-idf/components/wpa_supplicant/include -I/Users/adam/esp/esp-idf/components/wpa_supplicant/port/include -I/Users/adam/esp/esp-idf/components/app_trace/include -I/Users/adam/esp/esp-idf/components/app_update/include -I/Users/adam/esp/esp-idf/components/pthread/include -I/Users/adam/esp/esp-idf/components/smartconfig_ack/include -I/Users/adam/esp/esp-idf/components/sdmmc/include -I/Users/adam/esp/esp-idf/components/esp32-camera/driver/include -I/Users/adam/esp/esp-idf/components/esp32-camera/conversions/include -I/Users/adam/esp/esp-idf/components/esp_common/include -I/Users/adam/esp/esp-idf/components/esp_eth/include -I/Users/adam/esp/esp-idf/components/esp_event/private_include -I/Users/adam/esp/esp-idf/components/esp_rom/include -I/Users/adam/esp/esp-idf/components/esp_wifi/include -I/Users/adam/esp/esp-idf/components/esp_wifi/esp32/include -I/Users/adam/esp/esp-idf/components/lwip/include/apps/sntp -I/Users/adam/esp/esp-idf/components/spi_flash/private_include -I/Users/adam/esp/esp-idf/components/wpa_supplicant/include/esp_supplicant -I/Users/adam/esp/esp-idf/components/xtensa/include -I/Users/adam/esp/esp-idf/components/xtensa/esp32/include -I. -I -I/lib/mp-readline -I/lib/netutils -I/lib/timeutils -Ibuild -I/Users/adam/Downloads/pycparser-e1a1d737be66308b633215fa26ac5ed30e890103/utils/fake_libc_include -I/Users/adam/esp/xtensa-esp32-elf/xtensa-esp32-elf/include -I/Users/adam/Documents/Development/microPython/project/micropython -I/Users/adam/Documents/Development/microPython/project/micropython/ports/esp32/boards/GENERIC -I/Users/adam/Documents/Development/microPython/project/micropython/ports/esp32 -I/Users/adam/Documents/Development/microPython/project/micropython/ports/esp32/build-GENERIC ./test.h >> build/espidfmod/test.pp.c
python3 ../../gen/gen_mpy.py -I/Users/adam/esp/esp-idf/components/bootloader_support/include -I/Users/adam/esp/esp-idf/components/bootloader_support/include_bootloader -I/Users/adam/esp/esp-idf/components/console -I/Users/adam/esp/esp-idf/components/driver/include -I/Users/adam/esp/esp-idf/components/driver/include/driver -I/Users/adam/esp/esp-idf/components/efuse/include -I/Users/adam/esp/esp-idf/components/efuse/esp32/include -I/Users/adam/esp/esp-idf/components/esp32/include -I/Users/adam/esp/esp-idf/components/espcoredump/include -I/Users/adam/esp/esp-idf/components/soc/include -I/Users/adam/esp/esp-idf/components/soc/esp32/include -I/Users/adam/esp/esp-idf/components/heap/include -I/Users/adam/esp/esp-idf/components/log/include -I/Users/adam/esp/esp-idf/components/nvs_flash/include -I/Users/adam/esp/esp-idf/components/freertos/include -I/Users/adam/esp/esp-idf/components/esp_ringbuf/include -I/Users/adam/esp/esp-idf/components/esp_event/include -I/Users/adam/esp/esp-idf/components/tcpip_adapter/include -I/Users/adam/esp/esp-idf/components/lwip/lwip/src/include -I/Users/adam/esp/esp-idf/components/lwip/port/esp32/include -I/Users/adam/esp/esp-idf/components/lwip/include/apps -I/Users/adam/esp/esp-idf/components/lwip/include/apps/sntp -I/Users/adam/esp/esp-idf/components/mbedtls/mbedtls/include -I/Users/adam/esp/esp-idf/components/mbedtls/port/include -I/Users/adam/esp/esp-idf/components/mdns/include -I/Users/adam/esp/esp-idf/components/mdns/private_include -I/Users/adam/esp/esp-idf/components/spi_flash/include -I/Users/adam/esp/esp-idf/components/ulp/include -I/Users/adam/esp/esp-idf/components/vfs [... cut ...]
  File "/Users/adam/Library/Python/3.8/lib/python/site-packages/pycparser/c_parser.py", line 1858, in p_error
    self._parse_error(
  File "/Users/adam/Library/Python/3.8/lib/python/site-packages/pycparser/plyparser.py", line 67, in _parse_error
    raise ParseError("%s: %s" % (coord, msg))
pycparser.plyparser.ParseError: /Users/adam/esp/esp-idf/components/esp_common/include/esp_err.h:80:118: before: __attribute__
make: *** [build/espidfmod/test.c] Error 1`

I've removed the if statement to try to force it to define __attribute__ but am still getting the error.

adahl5, 2020-07-27:
Okay, I fixed that issue because I was making the wrong file. Now I am receiving the error: `Traceback (most recent call last):
  File "../../gen/gen_mpy.py", line 225, in <module>
    ast = parser.parse(s, filename='<none>')
  File "/Users/adam/Library/Python/3.8/lib/python/site-packages/pycparser/c_parser.py", line 149, in parse
    return self.cparser.parse(
  File "/Users/adam/Library/Python/3.8/lib/python/site-packages/pycparser/ply/yacc.py", line 331, in parse
    return self.parseopt_notrack(input, lexer, debug, tracking, tokenfunc)
  File "/Users/adam/Library/Python/3.8/lib/python/site-packages/pycparser/ply/yacc.py", line 1199, in parseopt_notrack
    tok = call_errorfunc(self.errorfunc, errtoken, self)
  File "/Users/adam/Library/Python/3.8/lib/python/site-packages/pycparser/ply/yacc.py", line 193, in call_errorfunc
    r = errorfunc(token)
  File "/Users/adam/Library/Python/3.8/lib/python/site-packages/pycparser/c_parser.py", line 1858, in p_error
    self._parse_error(
  File "/Users/adam/Library/Python/3.8/lib/python/site-packages/pycparser/plyparser.py", line 67, in _parse_error
    raise ParseError("%s: %s" % (coord, msg))
pycparser.plyparser.ParseError: /Users/adam/Documents/Development/microPython/project/micropython/ports/esp32/mpthreadport.h:37:5: before: SemaphoreHandle_t
make: *** [build/espidfmod/test.c] Error 1`

amirgon, 2020-07-27:
> Okay, I fixed that issue because I was making the wrong file. Now I am receiving the error:

Please also provide the command line, not only the error message.
Please specify the changes you did in the code/Makefiles.

adahl5, 2020-07-27:
I simply ran make, here is the makefile: [makefile](https://gist.github.com/adahl5/40db59015e682173e4452d523eac321d). Here is the header file: [file](https://gist.github.com/adahl5/1c49c0face8ccf39989876276f9e9493)

amirgon, 2020-07-28:
@adahl5 this problem is not related to macOS, as far as I can see.
You are trying to use `gen_mpy.py`  to process your own H file, and you are including many other headers (esp_system.h, esp_system.h etc.) which should not be processed by it on usual circumstances.

Apparently you are trying to expose some new API to micropython. (camera?)

`gen_mpy.py` is used to create Micropython API to specific libraries (lvgl, lodepng, and parts of esp-idf).
It can be used as a generic tool to create Micropython API to other libraries, but you can't just feed it any H file without being aware of its limitations.
For example, it can only process files written in [C99 language (according to the standard ISO/IEC 9899)](https://github.com/eliben/pycparser#13which-version-of-c-does-pycparser-support). It does not support gcc extensions or other extensions. It requires strict calling conventions in order to support callbacks.

So in general, you should have a pretty good understanding of how `gen_mpy.py` works in order to use it for your own library.

The simplest way to use it is to create a new H file with only the API you need and no includes to other H files.
You can add the includes on the C file where you implement your API. (`gen_mpy.py`  only processes the header files)  
If you want to use an existing API I recommend you wrap that API in such C functions.

More information is in the forum. For example here: https://forum.lvgl.io/t/how-crop-some-function-in-espidf-c-file/2775

**In general, GitHub issues are used to report bugs and this problem is not a bug but a "How To" question related to `gen_mpy.py`.
I suggest posting to [the forum](https://forum.lvgl.io/c/micropython/8) for questions like this. The forum reaches a wider audience and you'll have a better chance to get answers there.**

closed 2020-07-28 by amirgon