lvgl/lv_binding_micropython #92: Usage of ESP-IDF MDNS functions?

brotherdust, 2020-08-17:
Hello!
I notice that `mdns.h` is included in `espidf.h` and wanted to try it out. I'll admit that I'm not very literate with C, but through some poking around I was able to find the function I wanted, which was a simple mDNS query.

```
import espidf
ip_addr = espidf.ip_addr_t()
q = espidf.mdns_query_a("myhost.local", 2000, ip_addr)
```
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
SyntaxError: Can't convert ip_addr_t to ip4_addr_t!
```
Ok. I guess this was expected because I didn't hand it a `ip4_addr_t`. So, maybe it needs to be included in `espidf.h`? Ok. Yeah. Let's try that. I added `#include "lwip/ip4_addr.h"` to `espidf.h` and recompiled, expecting `gen_mpy` to take care of the rest, which it doesn't.

So, my questions are:

1. Why is `mdns.h` included in the first place?
2. Why isn't `gen_mpy` adding the types from `lwip/ip4_addr.h`?
3. Can you provide an example of using the ESP-IDF's mDNS functions, as provided by `gen_mpy`, in uPy? Maybe I'm just doing this all wrong.

Any help would be very much appreciated.
Thanks!

amirgon, 2020-08-17:
Hi @brotherdust, thank you for reporting this!
It was a bug in the Micropython binding script.

Fixed on master branch by 4e616d7090f101be28c5366785c427ed95012fb0.

This works now:
```python
ip_addr = espidf.ip4_addr_t()
q = espidf.mdns_query_a("myhost.local", 2000, ip_addr)
```

No need to make any changes in `espidf.h`.

closed 2020-08-17 by amirgon

brotherdust, 2020-08-17:
@amirgon thanks for the quick response!

`gen_mpy` is such a great tool! I really think it should be made a separate project so that it can be more generalized (and documented!). This would allow the ESP32 port of micropython to utilize features of the IDF almost as soon as they are released, rather than waiting for someone to implement by hand.

What say you?

brotherdust, 2020-08-18:
@amirgon 
Also, when I run the query, (as specified above), `ip_addr.p4addr_ntoa()` returns `0.0.0.0`, which indicates either a) I'm not setting up the pointer correctly, or b) the query fails. 

Interestingly, `espidf.esp_err_to_name(q)` returns `ESP_ERR_NOT_FOUND`. When I run a packet sniffer, I can see the mDNS query and the answer, but the ESP seems not to be listening for it.

Any thoughts? Maybe I'm not setting this up correctly. How would you go about doing a simple mDNS query?
Thanks!

amirgon, 2020-08-18:
> `gen_mpy` is such a great tool! I really think it should be made a separate project so that it can be more generalized (and documented!). This would allow the ESP32 port of micropython to utilize [...]

Thanks for this feedback @brotherdust !  

Originally I created the Micropython binding script for generating lvgl API, but later I've generalized it to handle other C libraries such as [lodepng](https://github.com/lvandeve/lodepng) and esp-idf.

This was a year ago. I've given it [some publicity](https://hackaday.com/2019/08/31/micropython-and-c-play-together-better/) and waited to see if I could identify any growing interest in the binding script outside the lvgl community.
Unfortunately I've seen relatively little public interest in adopting the Micropython binding script for esp-idf or other C libraries.  
So currently, making the binding script a project of its own is just not worth the effort for me. But it's still actively used and maintained in context of lvgl and can be used with other libraries if anyone is willing to try it.

By the way, you should be aware that there is another parallel effort on Micropython to provide direct access to esp-idf on https://github.com/micropython/micropython/pull/5711. It allows calling esp-idf functions but it might be a bit more difficult to work with esp-idf structs with that approach.

Another thing to consider is that there are some limitations to the binding script.
For example, it requires special calling conventions when working with callbacks (when the C API expects you to provide a function pointer).  
I'm working in collaboration with lvgl maintainers to make sure lvgl C API follows these conventions and therefore it's pretty simple to make a Micropython callable object invoked upon a button click on the GUI. 
But obviously I have no influence over esp-idf (or other) library APIs, and in many cases callbacks are not supported well over there. This can still be mitigated by creating a wrapping API but requires an extra effort.

> Also, when I run the query, (as specified above), `ip_addr.p4addr_ntoa()` returns `0.0.0.0`, which indicates either a) I'm not setting up the pointer correctly, or b) the query fails.

Actually your code is working perfectly for me (ESP32 WROVER board):

```python
>>> import espidf
>>> ip_addr = espidf.ip4_addr_t()
>>> q = espidf.mdns_query_a("mymachine", 2000, ip_addr)
>>> espidf.esp_err_to_name(q)
'ESP_OK'
>>> ip_addr.p4addr_ntoa()
'10.0.0.9'
```

- Are you working on STA mode?
- Did you first connect to an access point and verified there are no issues with the connection?

I suggest you try this on C first, I don't think this problem is related to the Micropython binding.

brotherdust, 2020-08-19:
@amirgon , yep, you were right. It wasn't the binding. It was a network problem. Recent firmware update on my switch broke multicast. THANK YOU for your help.