lvgl/lv_binding_micropython #93: Building with LV_COLOR_DEPTH=32 fails

persello, 2020-09-01:
Hello, I'm trying to build the Micropython firmware for an ESP32 project which uses an ILI9488 LCD display.
As written in `ili9XXX.py`, I need to compile with `LV_COLOR_DEPTH` set to 32. The build fails because the ILI9341 driver (which I don't use) wants it to be set to 16.

The error I get is:
```c
../../lib/lv_bindings/driver/esp32/modILI9341.c:19:2: error: #error "modILI9341: LV_COLOR_DEPTH must be set to 16!"
 #error "modILI9341: LV_COLOR_DEPTH must be set to 16!"
```

And I compile with:
```zsh
make -C ports/esp32 LV_CFLAGS="-DLV_COLOR_DEPTH=32" FLASH_SIZE=16MB BOARD=GENERIC V=1 PORT=/dev/cu.usbserial-A1076LH4 deploy -j 4
```

I'm on macOS Catalina and the build with `LV_COLOR_DEPTH` set to 16 succeeds, but obviously throws an error at runtime when trying to load the ILI9488 driver.

This is my first time with Micropython, so I might be doing something wrong.

amirgon, 2020-09-02:
Hi @persello 
You are right, the Makefile included the old C drivers (modILI9341.c) which is no longer supported.
I removed it on https://github.com/lvgl/lv_micropython/commit/83589fd1a163b9b2cbff28e12711fa1190e488ba

Please let me know if this solves your problem.

persello, 2020-09-02:
Compiled successfully, thanks.

closed 2020-09-02 by persello