lvgl/lv_binding_micropython #98: Build for ESP32 without SPIRAM fails

gersch07, 2020-10-11:
Dear all,

when I tried to build the most recent version of LVGL + Micropython for ESP32 without SPIRAM (board = GENERIC), the linking process stops with following error:
`LINK build-GENERIC/application.elf
xtensa-esp32-elf-ld: build-GENERIC/esp-idf/esp_tcp_transport/libesp_tcp_transport.a(transport_tcp.o):(.literal.tcp_close+0x0): undefined reference to `close'
xtensa-esp32-elf-ld: build-GENERIC/esp-idf/esp_tcp_transport/libesp_tcp_transport.a(transport_tcp.o):(.literal.tcp_write+0x0): undefined reference to `write'
xtensa-esp32-elf-ld: build-GENERIC/esp-idf/esp_tcp_transport/libesp_tcp_transport.a(transport_tcp.o):(.literal.tcp_read+0x0): undefined reference to `read'
xtensa-esp32-elf-ld: build-GENERIC/esp-idf/esp_tcp_transport/libesp_tcp_transport.a(transport_tcp.o): in function `tcp_close':
/home/jks/esp/esp-idf/components/tcp_transport/transport_tcp.c:167: undefined reference to `close'
xtensa-esp32-elf-ld: build-GENERIC/esp-idf/esp_tcp_transport/libesp_tcp_transport.a(transport_tcp.o): in function `tcp_write':
/home/jks/esp/esp-idf/components/tcp_transport/transport_tcp.c:99: undefined reference to `write'
xtensa-esp32-elf-ld: build-GENERIC/esp-idf/esp_tcp_transport/libesp_tcp_transport.a(transport_tcp.o): in function `tcp_read':
/home/jks/esp/esp-idf/components/tcp_transport/transport_tcp.c:109: undefined reference to `read'
xtensa-esp32-elf-ld: build-GENERIC/esp-idf/esp_tcp_transport/libesp_tcp_transport.a(transport_tcp.o): in function `setsockopt':
/home/jks/esp/esp-idf/components/lwip/lwip/src/include/lwip/sockets.h:634: undefined reference to `close'
xtensa-esp32-elf-ld: build-GENERIC/esp-idf/esp_tls/libesp_tls.a(esp_tls.o): in function `esp_tcp_connect':
/home/jks/esp/esp-idf/components/esp-tls/esp_tls.c:149: undefined reference to `close'
xtensa-esp32-elf-ld: build-GENERIC/esp-idf/esp_tls/libesp_tls.a(esp_tls.o): in function `esp_int_event_tracker_capture':
/home/jks/esp/esp-idf/components/esp-tls/private_include/esp_tls_error_capture_internal.h:52: undefined reference to `close'
xtensa-esp32-elf-ld: build-GENERIC/esp-idf/mbedtls/libmbedtls.a(net_sockets.o):(.literal.mbedtls_net_recv+0x4): undefined reference to `read'
xtensa-esp32-elf-ld: build-GENERIC/esp-idf/mbedtls/libmbedtls.a(net_sockets.o):(.literal.mbedtls_net_send+0x4): undefined reference to `write'
xtensa-esp32-elf-ld: build-GENERIC/esp-idf/mbedtls/libmbedtls.a(net_sockets.o):(.literal.mbedtls_net_free+0x4): undefined reference to `close'
xtensa-esp32-elf-ld: build-GENERIC/esp-idf/mbedtls/libmbedtls.a(net_sockets.o): in function `mbedtls_net_recv':
/home/jks/esp/esp-idf/components/mbedtls/port/net_sockets.c:334: undefined reference to `read'
xtensa-esp32-elf-ld: build-GENERIC/esp-idf/mbedtls/libmbedtls.a(net_sockets.o): in function `mbedtls_net_send':
/home/jks/esp/esp-idf/components/mbedtls/port/net_sockets.c:407: undefined reference to `write'
xtensa-esp32-elf-ld: build-GENERIC/esp-idf/mbedtls/libmbedtls.a(net_sockets.o): in function `mbedtls_net_free':
/home/jks/esp/esp-idf/components/mbedtls/port/net_sockets.c:438: undefined reference to `close' `

However, building for ESP32 WROVER (BOARD = GENERIC_SPIRAM) finishes without any error, just like standard Micropython does for both targets. Does anyone else have these issues?

amirgon, 2020-10-12:
Hi @gersch07 !
The problem is missing c library functions during link (close, write, read, etc.)  

It took me some time to figure this out, but apparently [libc-psram-workaround.a](https://github.com/espressif/esp-idf/blob/release/v4.0/components/newlib/lib/libc-psram-workaround.a) defines these functions while [libc.a](https://github.com/espressif/esp-idf/blob/release/v4.0/components/newlib/lib/libc.a) doesn't!
It definitely looks like a bug in ESP-IDF, probably fixed on more recent versions.

Fortunately `libg.a` contains these missing functions as well, so replacing libc.a with libg.a seems to solve the problem.

amirgon, 2020-10-12:
Fixed by https://github.com/lvgl/lv_micropython/commit/59334c073120de2fb9883654dbdc13c69482decf

closed 2020-10-12 by amirgon