lvgl/lv_binding_micropython #101: gcc compile error: invalid initializer

heyuanjie87, 2020-11-04:
STATIC inline lv_color32_t* mp_write_ptr_lv_color32_t(mp_obj_t self_in)
{
    mp_lv_struct_t *self = MP_OBJ_TO_PTR(cast(self_in, get_mp_lv_color32_t_type()));
    return (lv_color32_t*)self->data;
}

#define mp_write_lv_color32_t(struct_obj) *mp_write_ptr_lv_color32_t(struct_obj)

STATIC mp_obj_t mp_lv_img_buf_get_px_color(size_t mp_n_args, const mp_obj_t *mp_args)
{
    //...
    lv_color_t color = mp_write_lv_color32_t(mp_args[3]);
    //...
}

amirgon, 2020-11-04:
@heyuanjie87 **please provide more details!**
lv_micopython builds without errors.
What did you change to get an error?
Please provide the steps to reproduce the problem, and the full error log.

heyuanjie87, 2020-11-04:
"typedef lv_color16_t lv_color_t;" used in my config.

LOG: "lv_mpy.c:5690:43: error: invalid initializer
 #define mp_write_lv_color32_t(struct_obj) *mp_write_ptr_lv_color32_t(struct_obj)"

amirgon, 2020-11-04:
@heyuanjie87 you did not provide the steps to reproduce the problem.
What platform are you using? Are you using lv_micropython? Which port? What is the `make` command line?

You can't expect help if you can't provide instructions how others can reproduce your errors.

closed 2020-11-16 by heyuanjie87