lvgl/lv_binding_micropython #104: compilation error on esp32 version

uraich, 2020-11-18:
I see that you included sh2lib.c in the new esp32 version of lv_bindings. If I try to compile with the original Makefile I get unresolved references from sh2lib.
I added sh2lib.c for compilation in the Makefile but when doing so I get plenty of compilation errors:

In file included from ../../lib/lv_bindings/driver/esp32/sh2lib.c:30:
../../lib/lv_bindings/driver/esp32/sh2lib.h:21:27: error: conflicting types for 'nghttp2_nv'
 typedef struct nghttp2_nv nghttp2_nv;
                           ^~~~~~~~~~
In file included from ../../lib/lv_bindings/driver/esp32/sh2lib.c:28:
/opt/ucc/afnog/afnog-2020/esp32-cam/esp-idf-micropython/components/nghttp/nghttp2/lib/includes/nghttp2/nghttp2.h:546:3: note: previous declaration of 'nghttp2_nv' was here
 } nghttp2_nv;
   ^~~~~~~~~~
../../lib/lv_bindings/driver/esp32/sh2lib.c: In function 'callback_on_frame_send':
../../lib/lv_bindings/driver/esp32/sh2lib.c:159:37: error: initialization of 'const nghttp2_nv *' {aka 'const struct nghttp2_nv *'} from incompatible pointer type 'nghttp2_nv * const' {aka 'struct <anonymous> * const'} [-Werror=incompatible-pointer-types]
             const nghttp2_nv *nva = frame->headers.nva;

... and many more
Looks like an incompatibility between the definitions in sh2lib.h and nghttp2.h from the espidf library

amirgon, 2020-11-18:
Hi @uraich,

> I see that you included sh2lib.c in the new esp32 version of lv_bindings. If I try to compile with the original Makefile I get unresolved references from sh2lib.

I have added it, but I did not push an update to lv_micropython yet.
Updating lv_bindings alone doesn't break lv_micropython because lv_micropython still uses the previous version of lv_bindings (the lv_bindings git submodule hash is the previous version).

I already have it compiling and working on my side, but still need to run some tests.
I hope to update lv_micropython soon (today/tomorrow).

sh2lib would provide http/2 support in Micropython, and I'm planning to use it for push, grpc etc.

uraich, 2020-11-18:
Ok, thanks. Please let me know when I can give compilation of sh2lib another try.
I had downloaded lv_micropython with git clone and added lv_bindings and lvgl with git submodules. This is how I ran into the errors. However, removing #include "sh2lib.h" from espidf.h in lv_bindings/drivers/esp32 allowed me to compile without problems.

amirgon, 2020-11-18:
> I had downloaded lv_micropython with git clone and added lv_bindings and lvgl with git submodules.

If you run `git submodule update --recursive` in lv_micropython, it would pull the correct version of each submodule, and you would not need to edit `espidf.h`.

amirgon, 2020-11-18:
Fixed on latest lv_micropython.
sh2lib is available.

closed 2020-11-18 by amirgon

uraich, 2020-11-19:
Hi Amir,
This is just to confirm that sh2lib compiles fine now. Thanks!

amirgon, 2020-11-19:
> This is just to confirm that sh2lib compiles fine now. Thanks!

Yes it does.
If you see problems please let me know.