lvgl/lv_binding_micropython #105: STM32F746 build failed

demoCrash, 2020-11-19:
Hello there.
According to https://blog.lvgl.io/2020-08-10/stm32-microptyothon-and-lvgl blog.
When I run
make -j8  BOARD=STM32F7DISC MICROPY_PY_LVGL=1 deploy-stlink
I got some errors as follows:
[image]

amirgon, 2020-11-19:
@prolomb / @embeddedt  / @kisvegabor - are you able to reproduce this on your side?

demoCrash, 2020-11-19:
I did solve the problem.
Here the steps:
1. go to lv_micropython/lib/ , remove the lv_bindings folder.
2. download the https://github.com/lvgl/lv_binding_micropython and checkout the branch 'bugfix/static_members' 
3. update all submodules and rename the folder to lv_bindings.
4. run  make -j8 BOARD=STM32F7DISC MICROPY_PY_LVGL=1 deploy-stlink again. maybe there  a error called "strncpy"  duplicate . Then I remove port/stm32/Makefile this line 
LIB_SRC_C += $(shell find $(LVGL_BINDING_DIR)/driver/stm32/string1.c)
5. Now. it build success.
Also I test the some python code it runs ok.
[image]
awsome!!! lvgl!!!

embeddedt, 2020-11-19:
I can reproduce this on my end. There are several problems here:

* `strncpy` now exists in MicroPython itself, so string1.c is no longer necessary.
* The syntax issue as reported by @demoCrash.
* The binary is now too large to fit into the (1 megabyte!) flash of STM32F7.

amirgon, 2020-11-19:
> * The syntax issue as reported by @demoCrash.

Could you explain the problem?
I see that `STATIC` is defined to `static`.
What is the problem with this macro expansion?

> * The binary is now too large to fit into the (1 megabyte!) flash of STM32F7.

There is a separate issue to discuss this: https://github.com/lvgl/lv_binding_micropython/issues/81

> 2\. download the https://github.com/lvgl/lv_binding_micropython and checkout the branch 'bugfix/static_members'

`bugfix/static_members` branch was already merged to master, so there must be some recent commit that broke stm32.

@embeddedt in general, there is a problem with the stm32 port - we really test it very rarely.
The unix (Linux) port builds and runs as part of the CI on every commit.
The ESP32 is used by me so I check it with almost every change I do.
But apparently stm32 is not used as frequently.
A possible solution could be to build the stm32 port in GitHub Actions CI like the unix port (as we discussed [in the forum](https://forum.lvgl.io/t/firmware-bin-for-esp32-wrover-kit/3722/4?u=amirgon)).

embeddedt, 2020-11-19:
> What is the problem with this macro expansion?

I am not sure. What I know is that I see the same issue that @demoCrash sees with a fresh copy, so something has indeed changed in the way this file gets generated. I'll have to take a look at it in more detail when I get a chance.

embeddedt, 2020-11-21:
The build now succeeds and runs, although there is very little flash space remaining. #81 needs to be revisited ASAP.

closed 2020-11-21 by embeddedt

embeddedt, 2020-11-21:
CI has also been implemented for STM32 - it is now rebuilt every time a pull request is opened or a commit is pushed (same as the Unix port). Unfortunately we have no way of actually testing the built binary in CI at the moment, but this should still be a significant step forward.