lvgl/lv_binding_micropython #106: problem with reset on ili9XXX

uraich, 2020-11-20:
On my ili9341 display the reset line is connected to system reset. This means that I have to define rst=-1 in the initialisation of the driver. rst should be treated like backlight and power when initializing the chip

uraich, 2020-11-20:
If I run the code below twice, I get: 
```
E (229610) spi: SPI3 already claimed by spi master.
```

```
import lvgl as lv
import sys
import time

import lvesp32
import espidf as esp
from ili9XXX import *
from xpt2046 import xpt2046

# Initialize ILI9341 display
disp = ili9XXX(miso=19,mosi=23,clk=18, cs=26, dc=5, rst=-1, power=-1, backlight=-1, 

backlight_on=0, power_on=0,
               spihost=esp.VSPI_HOST, mhz=40, factor=4, hybrid=True, width=320, height=240,
               colormode=COLOR_MODE_BGR, rot=LANDSCAPE, invert=False, double_buffer=True, half_duplex=True, display_type=0)

# Initialize touch screen
touch = xpt2046(spihost=esp.VSPI_HOST)
```
If I try to deinit the driver I get:
```
ili9XXX initialization completed
Double buffer
Deinitializing ili9XXX..
CORRUPT HEAP: multi_heap.c:477 detected at 0x3ffd4fec
abort() was called at PC 0x4009451b on core 1
```
Initialization does not seem to be complete neither. If I run a test program using the driver from 
https://github.com/jeffmer/micropython-ili9341 and then an lvgl program with the ili9XXX driver, the GUI shows up on the display as expected. If I run the same program after a reset the screen stays white.

uraich, 2020-11-20:
Sorry, I got one thing wrong:
```
disp = ili9XXX(miso=19,mosi=23,clk=18, cs=26, dc=5, rst=-1, power=-1, backlight=-1,
```
should read 
disp = ili9341(miso=19,mosi=23,clk=18, cs=26, dc=5, rst=-1, power=-1, backlight=-1,
Like this the initialization is complete.
The deinit problem is still there however

amirgon, 2020-11-20:
Hi @uraich ,

> rst should be treated like backlight and power when initializing the chip

That makes sense, thanks for pointing this out!  
Fixed on b9fbdc555f20381229ed1473ac0218d6c1940962

> If I run the code below twice, I get:
> 
> ```
> E (229610) spi: SPI3 already claimed by spi master.
> ```

Obviously, if SPI is already allocated to the driver, trying to create another instance of the driver that tries to claim the same SPI would fail, so this is the expected behavior.

> If I try to deinit the driver I get:
> 
> ```
> ili9XXX initialization completed
> Double buffer
> Deinitializing ili9XXX..
> CORRUPT HEAP: multi_heap.c:477 detected at 0x3ffd4fec
> abort() was called at PC 0x4009451b on core 1
> ```

Actually I've seen this before.
The problem also appears in a different context, when trying to load the driver after soft-reset.
In both cases the result is a memory corruption. It is hard to debug this because the crash happens long after the actual memory corruption happens.  

Current workaround (which may or may not be applicable in your case) is to initialize the driver only once until hard-reset.  
When I have more time I would dive into this again, but until then any help debugging this would be appreciated!

uraich, 2020-11-21:
Hi Amir
Initializing the driver only once after reset is ok for me for the moment. I may try to figure out what is going on if I find the courage. First however I want to prepare a pull request for the Python version of the lv_examples. These will contain all the widget examples and the demo-widgets examples. 
Thanks to your help I now managed to get them all to work.
You can close this issue.

closed 2020-11-21 by amirgon