lvgl/lv_binding_micropython #109: LVGL and BLE on ESP32

lixas, 2020-11-25:
Hello

I have an issue, that i can not use BLE in my LVGL micropython application. As and example i took [advanced demo application](https://github.com/lvgl/lv_binding_micropython/blob/master/examples/advanced_demo.py) as my starting point, for slightly modified ILI9341 and XPT2046. Built an UI that i like, and then i tried to implement BLE scanning functionality. And i failed at this point...

I was able to narrow down my issue to bare minimum:

```
import ubluetooth
import lvesp32          # <-- This breaks BLE scanner irq

def bt_irq(event, data):
    print(event, data)

ble = ubluetooth.BLE()
ble.active(True)
ble.irq(handler=bt_irq)
ble.gap_scan(0, 30000, 30000)
```

if lvesp32 is commented out- everything works as expected and i do see BLE advertisement packets. If i import lvesp32 later, when my BLE scanner is running- it stops like in second or so and no more data is shown. Importing lvesp32 is enough to kill/hang/stop BLE scanning process.

What I do need in my final application is read BLE advertisements packets, extract data and show in on LVGL interface on ILI9341 display.

I built firmware on my own computer for generic ESP32 (no external PSRAM)
MicroPython v1.9.4-2805-g46e9073b2 on 2020-09-01; ESP32 module with ESP32

amirgon, 2020-11-25:
Hi @lixas !

Please see related threads on forums:
- https://forum.lvgl.io/t/ble-issue-with-xpt2046-py-touch-driver/2604
- https://forum.micropython.org/viewtopic.php?f=18&t=8068

The lvesp32 job is to call `lv_tick_inc` and `lv_task_handler` periodically, and it does it by [creating a timer](https://github.com/lvgl/lv_binding_micropython/blob/9f08d6f2d85ba09a9cfe8af083f680431bec55e2/driver/esp32/modlvesp32.c#L51) (xTimerCreate) and [scheduling a task](https://github.com/lvgl/lv_binding_micropython/blob/9f08d6f2d85ba09a9cfe8af083f680431bec55e2/driver/esp32/modlvesp32.c#L42) for Micropython.
For some reason this approach does not work well with BLE, I'm not sure why.

Possible workarounds are:
- Change [lvesp32 tick parameter](https://github.com/lvgl/lv_binding_micropython/blob/9f08d6f2d85ba09a9cfe8af083f680431bec55e2/driver/esp32/modlvesp32.c#L53) to a larger value.
- Avoid lvesp32 completely and instead  call periodically `lv_tick_inc` and `lv_task_handler`.
- If you are using uasyncio you could use [lv_async](https://github.com/lvgl/lv_binding_micropython/blob/9f08d6f2d85ba09a9cfe8af083f680431bec55e2/lib/async_utils.py#L17) which replaces lvesp32 and does not require a dedicated timer.

lixas, 2020-11-27:
Thanks you for suggestions I'll share outcome of my tests so far:
Parameter =2 works for minimal scenario, bet did not really work for my app. I've finished with parameter =8. So far so good- everything works. But of course, I had to disable animations and transitions- it became choppy...
Also, i understood how to use lv_tick_inc and lv_task_handler functions, so maybe some day later i'll make transition to get rid of lvesp32 and call those functions when needed. Thanks you once again for showing me right way

closed 2020-12-31 by stale[bot]