lvgl/lv_binding_micropython #117: LVGL update

embeddedt, 2021-01-30:
Hi @amirgon,

Is there a chance you could bump the LVGL submodule to 7.9.1? lvgl/lvgl#2017 is preventing the JS port CI from rebuilding.

The last commit message to update lvgl says 7.9.1 but it appears that the actual version in use is 7.9.0.

amirgon, 2021-01-31:
Sure!
I'll do that tomorrow.

amirgon, 2021-01-31:
Done!

closed 2021-01-31 by amirgon

embeddedt, 2021-02-01:
Thanks! The MicroPython simulator has been updated.

https://lvgl.github.io/lv_micropython/bundle_out/index.html

amirgon, 2021-02-01:
> The MicroPython simulator has been updated.

But not at https://sim.lvgl.io/v7/micropython/ports/javascript/bundle_out/index.html, over there it's still shows the old one.
Is there a a way to redirect sim.lvgl.io/v7 to https://lvgl.github.io/? Or deploy it automatically over there too?

embeddedt, 2021-02-01:
Correct. I haven't set up https://github.com/lvgl/sim to be updated automatically yet.