lvgl/lv_binding_micropython #118: Merge Micropython v1.14

amirgon, 2021-02-02:
https://github.com/micropython/micropython/releases/tag/v1.14

embeddedt, 2021-02-02:
> support added for C++ user modules on unix, stm32 and esp32.

Is this something we'd want to look into?

amirgon, 2021-02-02:
Only if for some reason we want to build C++ user modules.
Currently the binding generates C modules, so I don't think that would be useful for us at this point.

On the other hand, one thing that might be interesting is the new Raspberry Pi RP2040 Micropython port. There's a lot of fuss around that.  
Maybe someone would want to run Micropython+LVGL on that new MCU.

amirgon, 2021-02-04:
@embeddedt I've just merged Micropython v1.14 and probably messed up the js and stm32 ports.
I've tested the unix and esp32 ports and they are good, I'm not sure what is causing the failures on js and stm32.  

Could you have a look?

embeddedt, 2021-02-05:
STM32 is fixed; JS should be fixed shortly once it rebuilds.

amirgon, 2021-02-05:
Great! Thanks @embeddedt !
All looks good, so I'm closing.

closed 2021-02-05 by amirgon