lvgl/lv_binding_micropython #119: micropython lv_binding_micropython/examples/fb_test.py  #ImportError: no module named 'evdev'

china-zhangdj, 2021-02-03:
I compiled and installed lv_micropython on raspberrypi. When I executed "micropython lv_binding_micropython/examples/fb_test.py". I encountered an error: "ImportError: no module named'evdev'", but when I executed import evdev in the REPL, it did not Report an error, what caused this.

amirgon, 2021-02-03:
Hi @china-zhangdj, 
`evdev` is a frozen module.
There is a Micropython bug related to `sys.path` affecting importing of frozen modules: https://github.com/micropython/micropython/issues/6419

To work around this, you can try running Micropython without command line arguments, and then something like this:

```python
import sys
sys.path.append('lib/lv_bindings/examples')
import fb_test
```

china-zhangdj, 2021-02-05:
thank you wery much!

execute：
```python3
 import sys
sys.path.append("")
```
can fix it.

amirgon, 2021-02-05:
Yes adding sys.path.append("") is a simpler workaround, thank you for this suggestion.

I'm closing this issue.

closed 2021-02-05 by amirgon