lvgl/lv_binding_micropython #122: Enable Micropython CI workflows

amirgon, 2021-02-05:
Currently all Micropython's CI workflows were disabled on https://github.com/lvgl/lv_micropython/commit/43ec34e2760f58299a4f780b602cb335b757476a

With some work, at least some of them can be enabled on lv_micropython.

embeddedt, 2021-02-05:
It's probably not worth re-enabling the ports ones (I think ours effectively do the same thing), but the formatting and code size ones look useful.

amirgon, 2021-02-05:
> It's probably not worth re-enabling the ports ones

My thought was to enable the port ones, specifically ports we don't support yet.  
That way we would know it builds on other platforms, and be ready to the moment someone starts using lv_micropython with a new port.

amirgon, 2021-02-20:
CI for stm32/esp32/unix ports could also be useful for providing prebuilt firmware effortlessly, as was [requested before](https://forum.lvgl.io/t/relationship-of-lv-micropython-and-lv-port-esp32/4628).

embeddedt, 2021-02-20:
@amirgon If you have a moment, can you take a look at the ESP-related commands in [this Action](https://github.com/lvgl/lv_micropython/blob/master/.github/workflows/esp32_port.yml), and let me know if I missed something? I think I've followed all the steps correctly but it's not finding some header files.

```
ESPIDFMOD-GEN build-GENERIC_SPIRAM/espidfmod/mp_espidf.c
In file included from ../../lib/lv_bindings/driver/esp32/espidf.h:101:
/home/runner/work/lv_micropython/lv_micropython/esp-idf/components/esp32/include/esp_clk.h:1:2: warning: #warning esp_clk.h has been replaced by esp32/clk.h, please include esp32/clk.h instead [-Wcpp]
 #warning esp_clk.h has been replaced by esp32/clk.h, please include esp32/clk.h instead
  ^~~~~~~
In file included from ../../lib/lv_bindings/driver/esp32/espidf.h:104:
/home/runner/work/lv_micropython/lv_micropython/esp-idf/components/driver/include/driver/gpio.h:17:10: fatal error: esp_err.h: No such file or directory
 #include "esp_err.h"
          ^~~~~~~~~~~
```

amirgon, 2021-02-21:
@embeddedt my thinking was to enable [Micropython's CI](https://github.com/micropython/micropython/blob/master/.github/workflows/ports_esp32.yml) instead of creating a custom one.  
The advantage is that we would follow upstream more closely.  This is especially important now that ESP32 port on Micropython is migrating to CMake.  

What do you think?

embeddedt, 2021-02-21:
The problem is that MicroPython's CI is not building multiple boards (at least when I investigated it). It is just building enough to verify that compilation still succeeds in general. E.g. for STM32 it is only building pyboard variants.

Do you think it's better to modify their CI script to do that than to make one for our purposes?

amirgon, 2021-02-21:
> Do you think it's better to modify their CI script to do that than to make one for our purposes?

I don't know about STM32, but for ESP32 the CI is maintained and modified according to build changes.  
For example, when Micropython supports a different (or an additional) ESP-IDF version, this is reflected in CI changes.

See for example: https://github.com/micropython/micropython/commit/aa3d6b6aa5ec9fdf15b5c2d30a1c9f9c9064cd3c#diff-398f66844239bcfec34a0362acfee807d8d697c15490291b03bce4904182a45f

So my thinking was - instead of maintaining the CI script separately ourselves and follow build changes, let's reuse Micropython's CI. 
Instead of modifying their scripts, I thought of trying to call these scripts and add additional actions (such as `actions/upload-artifact@v2`) where needed.

Creating prebuilt firmware would be nice and useful but I think it's a secondary goal.

amirgon, 2021-07-17:
@kisvegabor, @embeddedt I've [enabled ESP32 workflow](https://github.com/lvgl/lv_micropython/pull/32) and I was wondering about [providing prebuilt firmware](https://github.com/lvgl/lv_binding_micropython/issues/122#issuecomment-782750121)

- Do we really want to provide prebuilt firmware? Or leave it to the users as we do today? Would we be able to provide good support for novice users who use prebuilt firmware? One of the advantages of leaving the build step to the users is that today we almost always interact with users who have at least some technical understanding.
- If we do want to provide prebuilt firmware, for which boards?
- Technically how do we create a prebuilt firmware workflow? Where would we publish the pre-built firmware? What would trigger the workflow?

What do you think?

embeddedt, 2021-07-17:
> Technically how do we create a prebuilt firmware workflow? Where would we publish the pre-built firmware? What would trigger the workflow?

Take a look at this example:

https://github.com/lvgl/lv_port_stm32f746_disco/blob/master/.github/workflows/stm32_port.yml

We can then use https://nightly.link/ to provide direct links instead of needing to browse to the action artifacts.

amirgon, 2021-07-17:
> Take a look at this example:

What triggers the workflow? Every push? that could create lots of artifacts. Or do we need some release schedule?
Also, I think that GitHub holds artifacts for [no longer than 90 days](https://docs.github.com/en/github/administering-a-repository/managing-repository-settings/configuring-the-retention-period-for-github-actions-artifacts-and-logs-in-your-repository), so I'm not sure about storing artifacts in GitHub.

closed 2021-10-14 by amirgon