lvgl/lv_binding_micropython #123: When I set LV_COLOR_DEPTH in lv_conf.h to 16, an error was reported during compilation.

china-zhangdj, 2021-02-06:
When I set LV_COLOR_DEPTH in lv_conf.h to 16, an error was reported during compilation.
···shell
../../lib/lv_bindings/lvgl/src/lv_draw/lv_draw_label.c: In function 'lv_draw_label':
../../lib/lv_bindings/lvgl/src/lv_draw/lv_draw_label.c:310:24: error: 'color' may be used uninitialized in this function [-Werror=maybe-uninitialized]
             lv_color_t color ;
```

embeddedt, 2021-02-06:
Are you using the latest version? Line 310 of the latest version [is blank](https://github.com/lvgl/lvgl/blob/5e8d053adf9dcd8ab007382a0773c6a67a97ada3/src/lv_draw/lv_draw_label.c#L310). It's possible the issue has already been fixed.

china-zhangdj, 2021-02-07:
Thank you, it has indeed been fixed in the new version.

closed 2021-02-07 by amirgon