lvgl/lv_binding_micropython #125: Compile error under lv_micropython: sh2lib.c

andrewleek-droplab, 2021-02-16:
Greetings ... Thanks for all your great work on this!

Im getting an error when trying to compile lv_micropython for an esp32.  The error im seeing seems to be in sh2lib.
Im compiling on OSX Mojave using IDF v3.3.2 (9e70825d1e1cbf7988cf36981774300066580ea7)

`CC ../../lib/utils/sys_stdio_mphal.c
CC ../../lib/lv_bindings/driver/esp32/modlvesp32.c
CC ../../lib/lv_bindings/driver/esp32/modrtch.c
CC ../../lib/lv_bindings/driver/esp32/espidf.c
CC ../../lib/lv_bindings/driver/esp32/sh2lib.c
../../lib/lv_bindings/driver/esp32/sh2lib.c: In function 'sh2lib_init_handle':
../../lib/lv_bindings/driver/esp32/sh2lib.c:289:25: error: implicit declaration of function 'esp_tls_init' [-Werror=implicit-function-declaration]
         hd->http2_tls = esp_tls_init();
                         ^
../../lib/lv_bindings/driver/esp32/sh2lib.c:289:23: error: assignment makes pointer from integer without a cast [-Werror=int-conversion]
         hd->http2_tls = esp_tls_init();
                       ^
../../lib/lv_bindings/driver/esp32/sh2lib.c: In function 'sh2lib_connect_task_function':
../../lib/lv_bindings/driver/esp32/sh2lib.c:317:15: error: implicit declaration of function 'esp_tls_conn_new_sync' [-Werror=implicit-function-declaration]
     int res = esp_tls_conn_new_sync(hd->hostname, strlen(hd->hostname), 443, hd->http2_tls_cfg, hd->http2_tls);
               ^
cc1: all warnings being treated as errors
make: *** [build-GENERIC_SPIRAM/lib/lv_bindings/driver/esp32/sh2lib.o] Error 1`

Please let me know if you need further info.

Thanks,
Andrew

amirgon, 2021-02-16:
Hi @andrewleek-droplab, thank you for reporting this!
sh2lib is indeed incompatible with IDF v3.
I'll disable it for IDF < v4.

closed 2021-02-16 by amirgon

andrewleek-droplab, 2021-02-17:
Ah ok I see, I didnt think about it being a v3 vs v4 issue.

Well Im just glad I could help out.
Thanks for all the hard work on this!

commit 1b7261a2 (2021-02-16): sh2lib: Disable when espidf < v4.

Fixes #125