lvgl/lv_binding_micropython #127: lv_micropython does not go well with ulab

retsyo, 2021-02-22:
I can compile [ulab](https://github.com/v923z/micropython-ulab) into [official micropython](https://github.com/micropython/micropython)'s `ports/windows` via the simple way stated in the document of [ulab](https://github.com/v923z/micropython-ulab) in MSYS2+Mingw64
```
cd /where/I/cloned/micropython
cd ports/windows
make USER_C_MODULES=/where/I/cloned/ulab all
```
Then
```
E:\test_micropython>micropython.exe
MicroPython cf6a015-dirty on 2021-02-19; win32 version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import ulab
>>>
```

however, for [lv_micropython](https://github.com/lvgl/lv_micropython) with [my patch to enable building in MSYS2+Mingw64 on windows](https://github.com/lvgl/lv_binding_micropython/pull/126)
```
cd /where/I/cloned/lv_micropython
cd ports/windows
make USER_C_MODULES=/where/I/cloned/ulab all
```
which says
```
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Including User C Module from /e/ulab/code
LVGL-GEN build/lvgl/lv_mpy.c
LODEPNG-GEN build/lodepng/mp_lodepng.c
cp ../../lib/lv_bindings/driver/png/lodepng/lodepng.cpp build/lodepng/lodepng.c
mkdir -p build/genhdr
mkdir -p build/build/lodepng/
mkdir -p build/build/lvgl/
mkdir -p build/code/
mkdir -p build/code/numpy/
mkdir -p build/code/numpy/approx/
mkdir -p build/code/numpy/compare/
mkdir -p build/code/numpy/fft/
mkdir -p build/code/numpy/filter/
mkdir -p build/code/numpy/linalg/
mkdir -p build/code/numpy/numerical/
mkdir -p build/code/numpy/poly/
mkdir -p build/code/numpy/stats/
mkdir -p build/code/numpy/transform/
mkdir -p build/code/numpy/vector/
mkdir -p build/code/scipy/
mkdir -p build/code/scipy/optimize/
mkdir -p build/code/scipy/signal/
mkdir -p build/code/scipy/special/
...
CC ../../lib/timeutils/timeutils.c
LINK micropython.exe
   text    data     bss     dec     hex filename
2788744   42988  161728 2993460  2dad34 micropython.exe
```
As we can read that `Including User C Module from /e/ulab/code`. However
```
G:\lv_micropython\ports\windows>micropython.exe
MicroPython a66af07-dirty on 2021-02-22; win32 version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import ulab
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'ulab'
>>>
```

In other word, `ulab` in not linked into `lv_micropython`. 

I don't know whether `ulab` can be compiled into `lv_micropython` on other OS, but I think the answer is "NO". So is there a way to fix it? Thanks

amirgon, 2021-02-26:
> In other word, `ulab` in not linked into `lv_micropython`.
> 
> I don't know whether `ulab` can be compiled into `lv_micropython` on other OS, but I think the answer is "NO". So is there a way to fix it?

Hi @retsyo,

I've just built the unix port of lv_micropython with ulab as you described (`USER_C_MODULES=/where/I/cloned/ulab`) and it works as expected. I don't see any problem, I can import `ulab` and use it.

I suspect that what you are seeing might be related to the Window port.
We actively support the unix, esp32 and stm32 ports. The Window port is not maintained in lv_micropython although I think someone tried it in the past. There just aren't many people running Micropython directly under Windows.

I suggest you try this with the unix port.  
If you don't have a Linux machine you could run Linux on a VM with VirtualBox or VMware player instead. It's pretty easy to install and set up.

rockindy, 2021-03-12:
Hi @retsyo,

I have tried to build windows port of lv_micropython with ulab and can use it.
Here is my command:
```
$ make clean
$ make CROSS_COMPILE=i686-w64-mingw32- USER_C_MODULES=/home/yuming/work/micropython-ulab MICROPY_PY_FFI=0 MICROPY_PY_BTREE=0
```

The first clean command is important, please do not ignore it.

amirgon, 2021-04-05:
Closing since we can't reproduce the problem.
Feel free to open if the problem is still there.

closed 2021-04-05 by amirgon