lvgl/lv_binding_micropython #136: v8: Test stm32 port

amirgon, 2021-04-22:
Hi @embeddedt 

Would you be able to try running Micropython+LVGL-v.8 on stm32?
I've just pushed [dev-8.0 branch](https://github.com/lvgl/lv_micropython/tree/dev-8.0) to lv_micropython with latest LVGL and latest Micropython.

So far I've migrated [advanced_demo.py](https://github.com/lvgl/lv_binding_micropython/blob/dev-8.0/examples/advanced_demo.py) and tested it on unix and esp32 ports.

embeddedt, 2021-04-23:
Sure!

I tried it quickly tonight; there are a few minor errors which I will send patches for shortly. It does load and show the initial screen without issues, but there is some corruption appearing on the display when I try to interact with it (seems to be an issue with partial refreshes). It's late now so I will take a look when I get a chance next.

embeddedt, 2021-06-04:
I will reopen this issue if something comes up but v8 should be working fine on STM32 with #137.

closed 2021-06-04 by embeddedt