lvgl/lv_binding_micropython #141: Various upstreams still point to littlevgl, also found in URLs

ropg, 2021-05-23:
### Perform all steps below and tick them with [x]
- [X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [X] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
When cloning, my github client wil warn me that the upstream set for submodules does not match where they came from, giving me littlevgl URLs instead.

### To Reproduce

One example of many, from `lv_micropython/.gitmodules`:

```text
[submodule "lib/lv_bindings"]
	path = lib/lv_bindings
	url = https://github.com/littlevgl/lv_bindings.git
```

### Expected behavior

No deprecated URLs

### Screenshots or video

Nah...

### Offer

If that's appreciated I can go through at least everything related to micropython and fix it all where appropriate in a single PR per repo. But I'd first want to know that I'm not messing with things that will be cleaned anyway, or ...

kisvegabor, 2021-05-23:
Thanks for pointing it out. I moved the issue to the `lv_binding_micropython` repo.

amirgon, 2021-05-26:
> ### Offer
> If that's appreciated I can go through at least everything related to micropython and fix it all where appropriate in a single PR per repo. But I'd first want to know that I'm not messing with things that will be cleaned anyway, or ...

I just ignored these warnings so far.  
If you are willing to fix this, a PR would be appreciated!

amirgon, 2021-05-27:
Fixed by https://github.com/lvgl/lv_micropython/pull/26

Thanks.

closed 2021-05-27 by amirgon