lvgl/lv_binding_micropython #142: ili9341 display deinit throws error

ropg, 2021-05-23:
I know nobody ever deinits a screen, but I just did:

```python
>>> disp.deinit()
Deinitializing ILI9341..
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ili9XXX.py", line 246, in deinit
AttributeError: 'lv_disp_drv_t' object has no attribute 'remove'
```

ropg, 2021-05-26:
I've tried some things to see if I could fix it, but I run into not really understanding the order of business as lvgl deals with a soft boot. I'd love to have a system that I can Ctrl-D soft-boot without it hard-resetting though... I'd be happy to help if someone can give me some pointers...

amirgon, 2021-05-26:
> I'd love to have a system that I can Ctrl-D soft-boot without it hard-resetting though... I'd be happy to help if someone can give me some pointers...

The plan was that soft-reset would trigger deinit automatically, that's what `lvesp32.cb_finalizer` about.
But actually it would be cleaner to get a callback upon soft-reset from Micropython itself.
Here is a related discussion: https://github.com/micropython/micropython/issues/5487

I think that LVGL itself could be de-initiazlied with `lv.deinit()` at least theoretically, but I'm not sure if anyone is really de-initializing it and re-initializing it in the real world.

Apart from LVGL, deinit also:
- Unregister SPI
- deinit lvesp32
- Free DMAable RAMs

but I think there were some issues there, can't exactly remember, maybe the SPI de-init didn't work well.

On v8 the lvesp32 might change to a Python module instead of C module, which could make things a bit easier.