lvgl/lv_binding_micropython #146: fatal error: host/ble_hs.h : No such file or directory  [[same for esp_nimble_hci.h]]

ropg, 2021-05-28:
1) I feel the lv_micropython repository needs issues turned on, so there's a place for people to report with things specific to it.

2) Lacking that place: I try to compile the esp32 port on the `dev-8.0` branch of that and get:

```
/Users/rop/code/esp/lv_micropython/extmod/nimble/modbluetooth_nimble.c:38:10: fatal error: host/ble_hs.h: No such file or directory
 #include "host/ble_hs.h"
          ^~~~~~~~~~~~~~~
compilation terminated.
/Users/rop/code/esp/lv_micropython/ports/esp32/mpnimbleport.c:35:10: fatal error: esp_nimble_hci.h: No such file or directory
 #include "esp_nimble_hci.h"
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
```

It's probably something silly, but I'd really appreciate any pointers on how to solve. Turning off bluetooth in the menuconfig does not help.

amirgon, 2021-05-28:
> 1. I feel the lv_micropython repository needs issues turned on, so there's a place for people to report with things specific to it.

Originally, lv_micropython was created as an example of how to use lv_micropython_bindings on a Micropython fork.
As such, we try to keep changes there as minimal as possible and we try to keep it in sync with Micropython upstream releases. We also try to add changes to lv_micropython_bindings instead of to lv_micropython, when possible. (for example we keep all drivers in lv_micropython_bindings, the ESP32 CMake functionality etc.)

Eventually it turned out many people are using lv_micropython directly and only a few use it as a reference to support LVGL on their own Micropython fork, but I still think it makes sense to maintain most changes in lv_micropython_bindings and to keep lv_micropython as close to Micropython upstream as possible.

For now, please open issues here on lv_binding_micropython while questions can go to the [forum](https://forum.lvgl.io/c/micropython/8).

> 2\. Lacking that place: I try to compile the esp32 port on the `dev-8.0` branch of that and get:

What ESP-IDF version are you using?  
I'm currently using v4.2 tag (c40f2590bf759ff60ef122afa79b4ec04e7633d2) and I'm not seeing this problem.
Micropython is probably not compatible with *all* ESP-IDF versions.

ropg, 2021-05-28:
I tried v4.1, v4.2 and v4.3 tags, and had same problem with all of them... I'll try starting from scratch tomorrow and going step by step.

amirgon, 2021-06-02:
Are you still facing this problem?
I've just cleaned and rebuilt lv_micropython with ESP-IDF v4.2 without any problem.

Maybe that's something in the environment?
 - Maybe you didn't update submodules correctly? 
 - What toolchain version are you using? (I'm using xtensa gcc 8.4.0)
 - What CMake version are you using? (I'm using cmake 3.18.4)

> /Users/rop/code/esp/lv_micropython/extmod/nimble/modbluetooth_nimble.c:38:10: fatal error: host/ble_hs.h: No such file or directory
>  #include "host/ble_hs.h"

Looking for `ble_hs.h` I can find it on my workspace in two places:
- Under lv_micropython: `lib/mynewt-nimble/nimble/host/include/host/ble_hs.h`
- Under espidf: `components/bt/host/nimble/nimble/nimble/host/include/host/ble_hs.h`

Can you find it on these locations on your workspace?  
If not, maybe you can try cloning a fresh workspace, updating submodules etc.

ropg, 2021-06-02:
Oh sorry, should have posted a followup and closed the issue: it fixed itself when I fixed my weird state-issue with my own fork that I could only solve by starting over...

closed 2021-06-02 by ropg

hmmkay, 2021-06-05:
> Originally, lv_micropython was created as an example of how to use lv_micropython_bindings on a Micropython fork.
> As such, we try to keep changes there as minimal as possible and we try to keep it [...]

Hi Amirgon,

Can I suggest that you include your two paragraphs on lv_micropython vs lv_micropython_bindings early on in the main README.md somewhere? As a new user to the project I found it surprisingly difficult to understand how all the projects relate, and this explanation is helpful. This is a useful nugget of orientation info that doesn't deserve to get lost in the comment thread of a closed issued :)

amirgon, 2021-06-06:
> Can I suggest that you include your two paragraphs on lv_micropython vs lv_micropython_bindings early on in the main README.md somewhere?

Done.