lvgl/lv_binding_micropython #151: Improve CI tests

amirgon, 2021-05-31:
Currently [CI test](https://github.com/lvgl/lvgl/blob/master/.github/workflows/build_micropython.yml) builds and runs advanced_demo.py

- Need to make sure that if advanced_demo throws an exception, the test stops with an error. See: https://github.com/lvgl/lvgl/issues/1763#issuecomment-849874710
- It would make sense to run additional examples

kisvegabor, 2021-06-01:
We can run all examples created by @uraich too.

uraich, 2021-06-01:
I know that lv_example_style_14 still segfaults. I am looking into this

amirgon, 2021-06-01:
> We can run all examples created by @uraich too.

Examples are part of LVGL repo now, right?
I think that first step would be to push the @uraich's examples into LVGL.

kisvegabor, 2021-06-01:
> Examples are part of LVGL repo now, right?

Yes.

> I think that first step would be to push the @uraich's examples into LVGL.

I agree, adding minor changes later is easier.

amirgon, 2021-07-02:
Done:
- Added `tests/run.sh` and `tests/run_test.py` that run all tests under `/examples` and send some events.
- Changed GitHub workflows of lvgl, lv_binding_micropython, lv_micropython to run `tests/run.sh`.

closed 2021-07-02 by amirgon