lvgl/lv_binding_micropython #155: I developed a new module according to SDL_mouse.c, which is used to implement input devices through the network.After I bind the touch event, I click the button and get a segfault.

china-zhangdj, 2021-06-03:
I developed a new module according to SDL_mouse.c, which is used to implement input devices through the network. When I am running, they are all normal. The button can be touched and the normal touch state changes, but when I pass the button through btn.set_event_cb (event_handle) After binding the event, click the button again, micropython has a segfault, what is the cause? (The same mpy code runs normally with the SDL module, and the event_handle runs successfully). Thank you for your help.
New module link
[modnid.c](https://github.com/china-zhangdj/lv_mpy_net_indev/blob/main/modnid.c)

amirgon, 2021-06-03:
Please run with debugger and attach stack trace.

amirgon, 2021-06-22:
Closing due to inactivity.
Please reopen if you have new inputs.

closed 2021-06-22 by amirgon