lvgl/lv_binding_micropython #167: Allow imagetools.py be used in the JS simulator

amirgon, 2021-07-20:
Today `imagetools.py` relies on `@micropython.native` which is not supported in JS.
Need to provide empty implementation for this decorator specifically for the JS port and see if it runs well enough in the simulator in non native mode.

Example of the problem: 
https://sim.lvgl.io/v8.1/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/e3f0b85623962c9ff7e0b58813d075050d772b26/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/e3f0b85623962c9ff7e0b58813d075050d772b26/examples/widgets/img/lv_example_img_1.py

embeddedt, 2021-07-20:
Personally, I think it would make more sense if `@micropython.native` and `@micropython.viper` were optimization hints, and a specific platform which did not support them would simply fall back to normal bytecode execution, so I'm in favor of your solution.

Are there other platforms with no/limited support for these decorators? I wonder if it would be worthwhile to suggest this type of fallback upstream, instead of trying to monkey-patch it in our fork.

amirgon, 2021-07-20:
> Are there other platforms with no/limited support for these decorators? I wonder if it would be worthwhile to suggest this type of fallback upstream, instead of trying to monkey-patch it in our fork.

I think most ports support native/viper decorators, and those who still don't eventually gets support. For example ESP32 got native/viper support [less than two years ago](https://github.com/micropython/micropython/pull/5082).
It makes lots of sense for microcontrollers, but I'm not sure about JS. Worth trying, I guess.

amirgon, 2021-07-20:
@embeddedt it's going to be harder than simply removing/ignoring a decorator:

https://github.com/lvgl/lv_binding_micropython/blob/772ac989ecfc16ddb4eba6bda99513155692fd81/lib/imagetools.py#L102-L115

Viper syntax is different than Python syntax. In python there is no `ptr32`.

So here are some options:
- Provide a dummy Viper decorator and implement `ptr32` in Python. This is going to be slow...
- Implement the Viper functions in C instead
- Build Micropython JS port with a different color mode that doesn't require this conversion from rgba8888 to bgra8888. I don't think this is possible today but maybe it's possible to support rgba8888 color mode in LVGL? Would it be useful for other purposes? It would probably require changes in the SDL driver as well.

What do you think?

amirgon, 2021-07-20:
Implemented JS specific `imagetools.py` under `driver/js`.

[Example](https://sim.lvgl.io/v8.1/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/e3f0b85623962c9ff7e0b58813d075050d772b26/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/e3f0b85623962c9ff7e0b58813d075050d772b26/examples/others/snapshot/lv_example_snapshot_1.py&script_direct=34d1e0d5addd2ae4c1321813b4a63f4b2e1a8b0b)

closed 2021-07-20 by amirgon