lvgl/lv_binding_micropython #181: LVGL bindings to other languages , like Lua ?

wendal, 2021-09-23:
"About this repo" : LVGL bindings to other languages

I think , Lua is another languages too, not only Python/MicroPython. 

Our team is working at lv_lua , which is the Lua binding of LVGL 7.11 https://github.com/openLuat/LuatOS/tree/master/components/lvgl .

So, are you(lvgl team) interested in accepting it as lvgl/lv_lua ?

amirgon, 2021-09-23:
> So, are you(lvgl team) interested in accepting it as lvgl/lv_lua ?

Please consider targeting the bindings to version 8. It has many advantages over version 7.

CC @kisvegabor

Dozingfiretruck, 2021-09-23:
> > So, are you(lvgl team) interested in accepting it as lvgl/lv_lua ?
> 
> Please consider targeting the bindings to version 8. It has many advantages over version 7.
> 
> CC @kisvegabor

Hi,Lua binding of LVGL 7.11 has been completed,Whether to consider accepting the 7.11 later upgrade to V8?

kisvegabor, 2021-09-27:
A Lua binding sounds great, but for maintenance reasons a v8 version would better for us. We'd like motivate people to move to v8. 

Do you plan to update to v8?

kisvegabor, 2021-09-28:
> Yes.Our team plans to upgrade to V8 later, 

Happy to hear that. I'm looking forward to see it. :slightly_smiling_face: