lvgl/lv_binding_micropython #182: Compile error while porting to nrf

andycktsai, 2021-10-01:
Enviroment:
lv_microPython: commit 654a3e94a2178f26ece2a0d1cc9ba367bf5575f7
lv_binding_micropython:  commit 53c64a5527b1efcbd35a200ad7ba85631774f71f
LinuxMint20.2 64bits
arm-none-eabi-gcc (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

**Condition:
1. cd ports/nrf
2. make submodules --> OK
3. make --> NG, error message:

CC ../../lib/lv_bindings/lvgl/src/core/lv_obj_style_gen.c
CC ../../lib/lv_bindings/lvgl/src/core/lv_disp.c
CC ../../lib/lv_bindings/lvgl/src/core/lv_obj_style.c
In file included from ../../lib/lv_bindings/lvgl/src/core/../misc/lv_gc.h:69,
                 from ../../lib/lv_bindings/lvgl/src/core/lv_obj_style.c:11:
../../lib/lv_bindings/lvgl/src/core/lv_obj_style.c: In function '_lv_obj_style_init':
../../py/mpstate.h:289:40: error: 'mp_state_vm_t' {aka 'struct _mp_state_vm_t'} has no member named '_lv_obj_style_trans_ll'
  289 | #define MP_STATE_VM(x) (mp_state_ctx.vm.x)
      |                                        ^
./mpconfigport.h:302:23: note: in expansion of macro 'MP_STATE_VM'
  302 | #define MP_STATE_PORT MP_STATE_VM
      |                       ^~~~~~~~~~~
.
.
._(Too long, omitted)_
.
.
make: *** [../../py/mkrules.mk:77: build-pca10040/lib/lv_bindings/lvgl/src/core/lv_obj_style.o] Error 1**

amirgon, 2021-10-01:
Hi @andycktsai !

Porting lv_micropython requires some changes in the makefile and configuration files under ports/nrf.  

Please have a look at other ports (unix, stm32, esp32) as an example.

andycktsai, 2021-10-05:
Hi @amirgon ,
Thanks a lot, I saw your modification for stm32, I will check it.

closed 2021-10-05 by andycktsai