lvgl/lv_binding_micropython #183: align, set_align, and set_style_align: only 2 work

jdtsmith, 2021-10-03:
The micropython bindings for object alignment include the three methods `align`, `set_align`, and `set_style_align`.  `align` and `set_style_align` work (the former with x,y offsets), but `set_align` does not function in a recent build, silently ignoring its inputs.

amirgon, 2021-10-03:
@jdtsmith - I cannot reproduce the problem you are seeing.

See [this example](https://sim.lvgl.io/v8.1/micropython/ports/javascript/index.html?script_direct=417c5390c2cee4a7523b7c844298d7b1beb5409e) that works as expected:
```python
scr = lv.obj()
btn = lv.btn(scr)
btn.set_align(lv.ALIGN.CENTER)
```

Could you send an example that shows the problem?

jdtsmith, 2021-10-04:
Hmm, I see that it does indeed work for some elements.  Here's a reproduction using a `textarea`, which is where I have noticed the issue:

```python
ta = lv.textarea(lv.scr_act())
ta.set_style_bg_color(lv.color_black(),0)
ta.set_height(lv.scr_act().get_height()//2)
ta.set_align(lv.ALIGN.BOTTOM_MID) # Doesn't work
ta.set_style_align(lv.ALIGN.BOTTOM_MID,0) # Works
ta.set_align(lv.ALIGN.TOP_MID) # Doesn't work
ta.align(lv.ALIGN.TOP_MID, 0, 0) # Works
```

amirgon, 2021-10-04:
That's because `lv_textarea_set_align` has a different meaning than `lv_obj_set_align`.

See https://forum.lvgl.io/t/spinbox-text-centering-not-working-as-intended-expected/6762?u=amirgon

jdtsmith, 2021-10-04:
Aha, thanks.  So it's an inconsistency in the LVGL API, which it looks like they have [now deprecated](https://docs.lvgl.io/master/widgets/core/textarea.html#_CPPv421lv_textarea_set_alignP8lv_obj_t15lv_text_align_t) (_"Deprecated: use the normal text_align style property instead"_).

 I'll patch around it.

closed 2021-10-04 by jdtsmith