lvgl/lv_binding_micropython #196: CI fails due to LVGL external libraries

amirgon, 2021-11-06:
CI Failure only on `lv_binding_micropython` CI: https://github.com/lvgl/lv_binding_micropython/runs/4122948432?check_suite_focus=true

External libraries (FreeType and Rlottie) are not linked [although they are supposed to](https://github.com/lvgl/lv_micropython/blob/efb8b0a3d9ef3cc83168c986248f19122bffad69/ports/unix/Makefile#L223-L237):
```make
# Additional optional libraries

RLOTTIE_CFLAGS_MOD :=  $(shell pkg-config --silence-errors --cflags rlottie)
ifneq ($(RLOTTIE_CFLAGS_MOD),)
CFLAGS_MOD += $(RLOTTIE_CFLAGS_MOD) -DMICROPY_RLOTTIE=1
RLOTTIE_LDFLAGS_MOD := $(shell pkg-config --silence-errors --libs   rlottie)
LDFLAGS_MOD += $(RLOTTIE_LDFLAGS_MOD)
endif

FREETYPE_CFLAGS_MOD :=  $(shell pkg-config --silence-errors --cflags freetype2)
ifneq ($(FREETYPE_CFLAGS_MOD),)
CFLAGS_MOD += $(FREETYPE_CFLAGS_MOD) -DMICROPY_FREETYPE=1
FREETYPE_LDFLAGS_MOD := $(shell pkg-config --silence-errors --libs   freetype2)
LDFLAGS_MOD += $(FREETYPE_LDFLAGS_MOD)
endif
```

The same problem does not happen on `lv_micropython` or on `LVGL` CI.

@embeddedt - 
Maybe you would have an idea: How is it possible that this problem happens only on `lv_binding_micropython` but not on `lv_micropython` or `LVGL` CI, although they are building the unix port in the same way?

embeddedt, 2021-11-06:
I'd have to investigate it further, but at first glance, the extra `LDFLAGS_MOD` argument [here](https://github.com/lvgl/lv_binding_micropython/blob/2c803d0086fa8daf107d817c8c624bd1e55f8b3c/.github/workflows/unix_port.yml#L36) looks suspicious. I didn't add that to the LVGL CI.

amirgon, 2021-11-06:
> I'd have to investigate it further, but at first glance, the extra `LDFLAGS_MOD` argument [here](https://github.com/lvgl/lv_binding_micropython/blob/2c803d0086fa8daf107d817c8c624bd1e55f8b3c/.github/workflows/unix_port.yml#L36) looks suspicious. I didn't add that to the LVGL CI.

You are right, of course! Thank you for pointing this out.

LDFLAGS_MOD on the make command line overrides LDFLAGS_MOD in the Makefile.  
These linker options were [added for debug when the run script crashed with SIGSEGV](https://github.com/lvgl/lv_binding_micropython/commit/bce6a67308163a6e2fb642483a6aec170522e07c), but I thing they are no longer needed.

Fixed by https://github.com/lvgl/lv_binding_micropython/commit/34706664d57bdedf071d2aceaf322a926a61b78e

closed 2021-11-06 by amirgon