lvgl/lv_binding_micropython #197: Support ESP32S3 IDF 4.4

Kevincoooool, 2021-11-20:
Hi，Can you support the newest esp32s3？

amirgon, 2021-11-20:
Hi @Kevincoooool 
Which display driver are you planning to use?
It seems that esp-idf API is different between ESP32 and ESP32S3. So even if we support it, I'm not sure [existing display drivers](https://github.com/lvgl/lv_binding_micropython/blob/master/driver/esp32/ili9XXX.py) would work.

Kevincoooool, 2021-11-21:
> Hi @Kevincoooool 
> Which display driver are you planning to use?
> It seems that esp-idf API is different between ESP32 and ESP32S3. So even if we support it, I'm not sure [existing display drivers](https://github.com/lvgl/lv_binding_micropython/blob/master/driver/esp32/ili9XXX.py) would work.

Hello, I've seen that the original warehouse of micropython already supports esp32s3. For the API part, just set DMA to auto during SPI initialization. Others are compatible. Can you merge the lvgl part in the master branch of the original warehouse? Thank you

amirgon, 2022-02-24:
Related discussion on https://github.com/lvgl/lv_binding_micropython/issues/208
Let's continue the discussion there.

closed 2022-02-24 by amirgon