lvgl/lv_binding_micropython #204: Wrong disp_spi_init() at  ST7789 on ili9XXX.py

11wanqin, 2022-01-26:
https://github.com/lvgl/lv_binding_micropython/blob/f6f75e41fdc45ecd6c860647c7451c38aef4918a/driver/esp32/ili9XXX.py#L199

the mode should be equal to 2 when use st7889.
"mode": 2,

amirgon, 2022-01-26:
@russhughes - Could you confirm this?

11wanqin, 2022-01-26:
my lcd is not working   , when mode  be equal to 0.
But it is work well, when i clone  the  ili9XXX.py  and make the mode  be equal to 2.

The following is my code:

from st7789d import st7789
import lvgl as lv
import machine

disp = st7789( miso=-1, mosi=23, clk=18, cs=26, dc=2, rst=4, backlight=5,width=240, height=240,backlight_on=0, double_buffer = False, factor=4,hybrid=False,initialize=True)

# Create a button with a label 

scr = lv.obj()
btn = lv.btn(scr)
btn.set_align(lv.ALIGN.CENTER)
label = lv.label(btn)
label.set_text('Hello world.') 
lv.scr_load(scr)

amirgon, 2022-06-24:
Thank's @russhughes for this information!   
I'll add a new `spi_mode` argument with a default value of `0`.

commit ad828bd5 (2022-06-24): ili9XXX.py: Add spimode argument

Fixes #204
closed 2022-06-24 by amirgon