lvgl/lv_binding_micropython #208: lv_espidf.c not ready for esp32-s2?

dotnfc, 2022-02-24:
trying to build lv_mpy for esp32-s2, but get some compiling & linking errors.

i have to make change, 

e.g. espidf.h 
```
enum {
    #if (CONFIG_IDF_TARGET_ESP32)
    ENUM_SPI_HOST = SPI_HOST,
    ENUM_HSPI_HOST = HSPI_HOST,
    ENUM_VSPI_HOST = VSPI_HOST,
    #elif CONFIG_IDF_TARGET_ESP32S2
    ENUM_SPI_HOST = SPI_HOST,
    ENUM_HSPI_HOST = HSPI_HOST,
    ENUM_VSPI_HOST = FSPI_HOST,
    #endif // 
};
#endif

```

modrtch.c
```
#define GPIO_TO_ADC_ELEMENT(x) [x] = CONCAT3(ADC1_GPIO, x, _CHANNEL)

static const int gpio_to_adc[] = {
#if (CONFIG_IDF_TARGET_ESP32)
        GPIO_TO_ADC_ELEMENT(36),
        GPIO_TO_ADC_ELEMENT(37),
        GPIO_TO_ADC_ELEMENT(38),
        GPIO_TO_ADC_ELEMENT(39),
        GPIO_TO_ADC_ELEMENT(32),
        GPIO_TO_ADC_ELEMENT(33),
        GPIO_TO_ADC_ELEMENT(34),
        GPIO_TO_ADC_ELEMENT(35),
#elif (CONFIG_IDF_TARGET_ESP32S2 || CONFIG_IDF_TARGET_ESP32S3)
        GPIO_TO_ADC_ELEMENT(1),
        GPIO_TO_ADC_ELEMENT(2),
        GPIO_TO_ADC_ELEMENT(3),
        GPIO_TO_ADC_ELEMENT(4),
        GPIO_TO_ADC_ELEMENT(5),
        GPIO_TO_ADC_ELEMENT(6),
        GPIO_TO_ADC_ELEMENT(7),
        GPIO_TO_ADC_ELEMENT(8),
        GPIO_TO_ADC_ELEMENT(9),
        GPIO_TO_ADC_ELEMENT(10),
#elif (CONFIG_IDF_TARGET_ESP32C3)
        GPIO_TO_ADC_ELEMENT(1),
        GPIO_TO_ADC_ELEMENT(2),
        GPIO_TO_ADC_ELEMENT(3),
        GPIO_TO_ADC_ELEMENT(4),
#endif 
};
```

but , when linking, 

```
undefined reference to `rtc_gpio_force_hold_all' in function `gpio_force_hold_all'
undefined reference to `esp_eth_phy_new_lan87xx' in function `esp_eth_phy_new_lan8720':
undefined reference to `esp_eth_phy_new_lan87xx'
undefined reference to `esp_eth_phy_new_ksz8081'
undefined reference to `esp_eth_phy_new_ksz8041'
undefined reference to `esp_eth_phy_new_dp83848'
undefined reference to `esp_eth_phy_new_lan87xx'
undefined reference to `esp_eth_phy_new_rtl8201'
undefined reference to `esp_eth_phy_new_ip101'
```

theose eth_pyy_xxx may not be available for s2 chip, and the gpio_force_hold_all() in {esp-idf}\components\driver\gpio.c:

```
#if SOC_GPIO_SUPPORT_FORCE_HOLD

esp_err_t gpio_force_hold_all()
{
#if SOC_RTCIO_HOLD_SUPPORTED
    rtc_gpio_force_hold_all();
#endif
... ...

#endif // SOC_GPIO_SUPPORT_FORCE_HOLD
```

it should not be put in 'lv_espidf.c.pp.filtered' for esp32-s2, rigth? 

any suggestion?

amirgon, 2022-02-24:
> trying to build lv_mpy for esp32-s2, but get some compiling & linking errors.

We don't currently support esp32-s2, but I'll be happy to help any community-driven effort to add this support.

> theose eth_pyy_xxx may not be available for s2 chip, and the gpio_force_hold_all() in {esp-idf}\components\driver\gpio.c:

The esp-idf micropython module is auto-generated and contain a lot of bindings to functions (or structs) we don't really need. I don't see why we need `esp_eth_phy_*` or anything that rely on them, for example.

The automatic binding script scans header files and binds everything by default, **but we can exclude specific headers**.  
There are two ways to exclude headers:
- [With macros](https://github.com/lvgl/lv_binding_micropython/blob/81dadf150f12620240e970fe477df6591b0180de/driver/esp32/espidf.h#L12-L39) that define include guards (or define missing macros from excluded headers)
- [Filter headers in cmake](https://github.com/lvgl/lv_binding_micropython/blob/81dadf150f12620240e970fe477df6591b0180de/mkrules.cmake#L141-L149). This directly affects the generated `lv_espidf.c.pp.filtered`.

So I suggest you try to filter irrelevant headers. This would also reduce program size since the binding code would be smaller.  
When this works for you, we'll appreciate a PR to lv_binding_micropython!

dotnfc, 2022-02-24:
thanks for your comments, i'll try for get esp32-s2 supported.

dotnfc, 2022-02-28:
to check SDL is running here is a simple patch for modSDL.c

```
 driver/SDL/modSDL.c | 7 +++++++
 lvgl                | 0
 pycparser           | 0
 tests/run.sh        | 0
 4 files changed, 7 insertions(+)

diff --git a/driver/SDL/modSDL.c b/driver/SDL/modSDL.c
index 95a2a86..ff9eaf0 100644
--- a/driver/SDL/modSDL.c
+++ b/driver/SDL/modSDL.c
@@ -136,6 +136,12 @@ STATIC mp_obj_t mp_deinit_SDL()
     return mp_const_none;
 }
 
+STATIC mp_obj_t mp_in_service_SDL() 
+{
+    return mp_obj_new_bool(monitor_active());
+}
+
+MP_DEFINE_CONST_FUN_OBJ_0(mp_in_service_SDL_obj, mp_in_service_SDL);
 STATIC MP_DEFINE_CONST_FUN_OBJ_KW(mp_init_SDL_obj, 0, mp_init_SDL);
 STATIC MP_DEFINE_CONST_FUN_OBJ_0(mp_deinit_SDL_obj, mp_deinit_SDL);
 STATIC MP_DEFINE_CONST_FUN_OBJ_0(mp_refresh_SDL_obj, mp_refresh_SDL);
@@ -150,6 +156,7 @@ STATIC const mp_rom_map_elem_t SDL_globals_table[] = {
         { MP_ROM_QSTR(MP_QSTR_refresh), MP_ROM_PTR(&mp_refresh_SDL_obj) },
         { MP_ROM_QSTR(MP_QSTR_monitor_flush), MP_ROM_PTR(&PTR_OBJ(monitor_flush))},
         { MP_ROM_QSTR(MP_QSTR_mouse_read), MP_ROM_PTR(&PTR_OBJ(mouse_read))},
+        { MP_ROM_QSTR(MP_QSTR_in_service), MP_ROM_PTR(&mp_in_service_SDL_obj) },
 };
          
 
diff --git a/lvgl b/lvgl
--- a/lvgl
+++ b/lvgl
@@ -1 +1 @@
-Subproject commit 4bd1e7e9f7acc5295b65440477e76a048094afbf
+Subproject commit 4bd1e7e9f7acc5295b65440477e76a048094afbf-dirty
diff --git a/pycparser b/pycparser
--- a/pycparser
+++ b/pycparser
@@ -1 +1 @@
-Subproject commit 1706a39e0116dde0b2d1c52d67078a9a0ab4dbe7
+Subproject commit 1706a39e0116dde0b2d1c52d67078a9a0ab4dbe7-dirty
diff --git a/tests/run.sh b/tests/run.sh
old mode 100755
new mode 100644
```

and test code like this:

```
if __name__ == '__main__':
   if (sys.platform == "win32"):
       import SDL
       while SDL.in_service():          
          pass
```

dotnfc, 2022-03-01:
those esp_eth_phy_xxx symbols may be imported by 'mdns.h', but if i move the #include 'mdns.h' to espidf.c, failed on compiling frozen_contents.c

```
frozen_content.c:15963:17: error: 'MP_QSTR_ip' undeclared here (not in a function);
```
this is uftpd.py stuff
```
def num_ip(ip):
    items = ip.split(".") 
    ...
```

so complicated. ^^

amirgon, 2022-03-01:
> but if i move the #include 'mdns.h' to espidf.c

Don't remove 'mdns.h', it will remove functionality.
You can find out which headers define esp_eth_phy_xxx and add them to the FILTER exclusion list in mkrules.cmake. 

> failed on compiling frozen_contents.c

uftpd.py is not dependant on that. Probably a build dependencies problem, try to clean and rebuild.

kdmukai, 2022-07-24:
@amirgon against the current master (65304b6), I'm trying:
```
make BOARD=GENERIC_S2
```

My compilation gets further with @dotnfc's steps in the first post above, but now I'm getting stopped at:
```
modrtch.c:409:23: error: 'ADC_WIDTH_BIT_12' undeclared (first use in this function); did you mean 'ADC_WIDTH_BIT_13'?
     adc1_config_width(ADC_WIDTH_BIT_12);
                       ^~~~~~~~~~~~~~~~
                       ADC_WIDTH_BIT_13
```

I'm strictly a python guy so figuring out a fix from here is beyond me, unfortunately.

amirgon, 2022-07-25:
> I'm strictly a python guy so figuring out a fix from here is beyond me, unfortunately.

The S2 board, apparently, has different ESP-IDF API than the standard ESP32.  
At the moment I don't have the time to study the differences and put the effort to support other ESP32 variations such as S2.  
If someone from the community would like to volunteer, I would be happy to review a PR for that.   

But this would require come C knowledge and the willingness to spend time on this, read ESP documentation, etc.

kdmukai, 2022-07-26:
Understood. I'm rallying more C-savvy folks to try to take a look.

If you had to blindly guess, does this seem like a couple hours of work or are we more in the days or weeks range?

amirgon, 2022-07-26:
> If you had to blindly guess, does this seem like a couple hours of work or are we more in the days or weeks range?

It may be just a couple of hours, but I really can't tell without looking into it.  
And obviously, you'd need an S2 development board, preferably with JTAG so you could debug it.

amirgon, 2022-07-29:
Closing in favor of https://github.com/lvgl/lv_binding_micropython/issues/227

closed 2022-07-29 by amirgon