lvgl/lv_binding_micropython #209: fatal error: db.h: No such file or directory

brotherYoung, 2022-03-03:
/home/bmp/lv_micropython/extmod/modbtree.c:37:10: fatal error: db.h: No such file or directory
 #include <db.h>
          ^~~~~~
compilation terminated.
Traceback (most recent call last):
  File "/home/bmp/lv_micropython/py/makeqstrdefs.py", line 188, in <module>
    preprocess()
  File "/home/bmp/lv_micropython/py/makeqstrdefs.py", line 62, in preprocess
    for output in p.imap(pp(flags), chunks):
  File "/usr/lib/python3.8/multiprocessing/pool.py", line 868, in next
    raise value
  File "/usr/lib/python3.8/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/home/bmp/lv_micropython/py/makeqstrdefs.py", line 46, in run
    return subprocess.check_output(args.pp + flags + files)
  File "/usr/lib/python3.8/subprocess.py", line 415, in check_output
    returworks fine with micropython (without lv_bindings)n run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.8/subprocess.py", line 516, in run

it works fine with micropython。
OS: ubuntu 20.04 LTS (WSL)
idf: v4.2

looking forward to your reply,thanks.

amirgon, 2022-03-03:
Please send questions to the forum.  
https://forum.lvgl.io/c/micropython/8

closed 2022-03-03 by amirgon

wuyue92tree, 2022-04-11:
> 

I got this too. `make -C ports/esp32 submodules` is work for me.