lvgl/lv_binding_micropython #212: Can not build micropython with lvgl for `BOARD=STM32F769DISC`

r00t1900, 2022-05-25:
I've post this issue on https://github.com/micropython/micropython/issues/8696 and then I was informed that the error occurs on LVGL part.

In short, the lv_micropython firmaware for STM32F769-DISC can not compile on ubuntu18.04. As the repository https://github.com/lvgl/lv_micropython can not start an issue, I place the issue here.

Please help and find out why I can not build this firmware, thank you very much.

amirgon, 2022-05-25:
Hi @r00t1900 

I think that currently we only support STM32F7DISC, not STM32F769DISC, but I guess it should be pretty easy to add more boards.  
@embeddedt - what do you think?

embeddedt, 2022-05-27:
That's correct. We could probably fix the build error easily, but the port will still not be very useful without some work to adapt the F769 display and touchscreen drivers to a MicroPython module, as was done for [F746](https://github.com/lvgl/lv_binding_micropython/tree/master/driver/stm32/STM32F7DISC).

amirgon, 2022-05-27:
@r00t1900 - would you like to try and create a F769 display driver for lv_micropython?

r00t1900, 2022-05-28:
yup I'd like to try it even I am a beginner. I will try :)

amirgon, 2022-05-28:
> yup I'd like to try it even I am a beginner. I will try :)

Let us know if you need any help or advice!

closed 2022-10-21 by amirgon