lvgl/lv_binding_micropython #213: When MPY soft reboot and init lvgl again will halt in LV_ASSERT_HANDLER

imliubo, 2022-05-26:
Description
    When esp32 power on and init lvgl onece it works well, but when use "CTRL D" command try soft reboot MPY and init lvgl again will halt in **lv_style.c line #190**, which means it wiil stop at **LV_ASSERT_HANDLER**(define in lv_conf.h), can we avoid this? or if don't want halt here we can only change **LV_ASSERT_HANDLER** from **while(1)** to **esp_restart()** ?

I don't want restart esp32, so i'm here want ask some solution :), if I missed some things please tell me.

Full log
```bash
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4344
load:0x40078000,len:13816
load:0x40080400,len:3340
entry 0x40080618

MicroPython d0a9188-dirty on 2022-05-26; M5STACK with ESP32
Type "help()" for more information.
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import sys
=== import m5
=== from m5 import lcd
=== import lvgl as lv
=== from imagetools import get_png_info, open_png
=== 
=== # M5 init auto detect board type
=== m5.begin()
=== lcd.clear(0xff)
=== # lvgl init
=== lv.init()
[Info]  (0.000, +0)      lv_init: begin         (in lv_obj.c line #103)
>>> 
MPY: soft reboot
[Info]  (0.000, +0)      lv_deinit: lv_deinit done      (in lv_obj.c line #200)
[Info]  (0.000, +0)      lv_deinit: lv_deinit done      (in lv_obj.c line #200)

MicroPython d0a9188-dirty on 2022-05-26; M5STACK with ESP32
Type "help()" for more information.
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import sys
=== import m5
=== from m5 import lcd
=== import lvgl as lv
=== from imagetools import get_png_info, open_png
=== 
=== m5.begin()
=== lcd.clear(0xff)
=== # lvgl init
=== lv.init()
[Info]  (0.000, +0)      lv_init: begin         (in lv_obj.c line #103)
[Error] (0.000, +0)      lv_style_register_prop: Asserted at expression: MP_STATE_PORT(_lv_style_custom_prop_flag_lookup_table) != NULL (NULL pointer)       (in lv_style.c line #190)
```

amirgon, 2022-05-27:
Hi @imliubo 

Soft reset affects Micropython, but does not affect LVGL core, LVGL drivers and the hardware you are using.
For soft reset to work correctly, you must first de-init LVGL, the LVGL drivers you are using and possibly the hardware.  

My suggestion is to use hard reset instead (`machine.reset()`)

amirgon, 2022-05-28:
Fixed by https://github.com/lvgl/lvgl/commit/6a8dd1a3b64ab62a7685f526e1b588334b7324d6
But this only fixes the assert you are experiencing on lv_style.c.  You still need to de-init your m5/lcd driver first.

closed 2022-05-28 by amirgon

imliubo, 2022-05-30:
Confirmed, it was been fixed, thanks @amirgon !