lvgl/lv_binding_micropython #218: ports/esp32  stops building M1

Damon3499, 2022-06-23:
I got this repo lv_micropython downloaded yesterday. I use IDF 4.4 I don't switch the mircopython version. I figured out that the just Micropython from the Micropython version 1.19 build with no errors. I checked the lines in the fix you sent me, that's okay. I don't need to change it. Maybe this could help, when I call make V=1 -C mpy-cross.
[terminal.txt](https://github.com/lvgl/lv_binding_micropython/files/8964931/terminal.txt)

Damon3499, 2022-06-24:
Any idea?

amirgon, 2022-06-24:
Hi @Damon3499 
I'm not sure I understand what you are doing and what is the problem you are seeing. Your issue is not very clear.

Anyway:
- We currently support building lv_micropython only on Linux. I don't have a Mac and I can't support building it on Mac.
- Please follow the [build instructions in the Readme](https://github.com/lvgl/lv_micropython#build-instructions) to build it on Linux.
- lv_micropython is aligned to Micropython v1.18. (We plan to [migrate to v1.19](https://github.com/lvgl/lv_binding_micropython/issues/217) soon).
- v1.18 supports ESP-IDF v4.0.2 and 142bb32c50fa9875b8b69fa539a2d59559460d72 (commit before v5.0-dev). [See the CI script](https://github.com/lvgl/lv_micropython/blob/53e528dfb8cfdf63a09edb3f448ef139d2c4c6fb/tools/ci.sh#L105-L112)

closed 2022-06-24 by amirgon