lvgl/lv_binding_micropython #220: No Image Workshop on ESP 32

Damon3499, 2022-07-05:
Hello, I switched to Unbutu. Now I can build the image without any error. But the Images don‘t work on the esp. I tried almost every Board. It is rebooting all the time.  I tried M5Stack Core 2, ESP32  dev kitv4, TTGo. What am I doing wrong? 
When I build the micropython repo everything is Fine. Could you please help me?

amirgon, 2022-07-05:
Please provide more information.

- Specify exactly how you built it
- Specify exactly on which board you are flashing the firmware
- Specify exactly what is displayed on UART before it's rebooting. Does it display some "Guru Meditation Error"? Backtrace?

Damon3499, 2022-07-06:
I have Unbutu 22.04 installed.

I've downloaded this repo: git clone --recursive https://github.com/lvgl/lv_micropython.git
I tried image on :
ESP - WROOM 32
M5 Stack core 2
TTGO
I do:
/lv_micropython  source tools/ci.sh && ci_esp32_idf44_setup
throw an error -> tool Xtensa -esp32-elf does not have a version compatible with platform linux-arm64
(IDF version v4.4 -dev-3513-g142bb32c50)
so I used the 5.0 version:
/lv_micropython  source tools/ci.sh && ci_esp32_ build
throw an error -> Failed to resolve component "tcpip_adapter"
so I used the 4.4 version:
/lv_micropython  source tools/ci.sh && ci_esp32_ build
throw an error -> Failed ninja

so I tried like in the Readme with idf 4.4
and can compile without error Xtensa esp-2021r2-8.4.0
Esp - WROOM32
/lv_micropython/ports/esp32 make erase
/lv_micropython/ make -C mpy-cross
/lv_micropython/ make  -C ports/esp32 LV_CFLAGS="-DLV_COLOR_DEPTH=16 -DLV_COLOR_16_SWAP=1" BOARD=GENERIC_SPIRAM deploy
the build runs without any errors -> mount the image on esp - WROOM 32 
-> error on WROOM32 
 E(874) psram: PSRAM ID read error: 0xffffffff
E(874) spiram: SPI RAM enabled but initialization failed. Bailing out.
Guru Meditation Error: Core 0' panic'ed (StoreProhibited). Exeption was unhandled.
Backtrace: 0x4019e177:0x3ffd45b00x400d887e:0x3ffd45e0 0x400d86f9:0x3ffd4610

TTGO
/lv_micropython/ports/esp32 make erase
/lv_micropython/ make  -C ports/esp32 LV_CFLAGS="-DLV_COLOR_DEPTH=16 -DLV_COLOR_16_SWAP=1" BOARD=GENERIC_SPIRAM deploy
he build runs without any errors -> mount the image on esp - TTGO
-> error on TTGO
 E(874) psram: PSRAM ID read error: 0xffffffff
E(874) spiram: SPI RAM enabled but initialization failed. Bailing out.
Guru Meditation Error: Core 0' panic'ed (StoreProhibited). Exeption was unhandled.
Backtrace: 0x4019e177:0x3ffd45b00x400d887e:0x3ffd45e0 0x400d86f9:0x3ffd4610

WROOM 32
/lv_micropython/ports/esp32 make erase
/lv_micropython/ make  -C ports/esp32 LV_CFLAGS="-DLV_COLOR_DEPTH=16 -DLV_COLOR_16_SWAP=1" BOARD=GENERIC deploy
the build runs without any errors -> mount the image on esp - WROOM 32 
-> error on WROOM32 
Guru Meditation Error: Core 0' panic'ed (StoreProhibited). Exeption was unhandled.
Backtrace: 0x4019372f:0x3ffd3ed00x400d8312:0x3ffd3f00 0x400d81a0:0x3ffd3f30

TTGO
/lv_micropython/ports/esp32 make erase
/lv_micropython/ make  -C ports/esp32 LV_CFLAGS="-DLV_COLOR_DEPTH=16 -DLV_COLOR_16_SWAP=1" BOARD=GENERIC deploy
the build runs without any errors -> mount the image on esp - WROOM 32 
-> error on TTGO
Guru Meditation Error: Core 0' panic'ed (StoreProhibited). Exeption was unhandled.
Backtrace: 0x4019372f:0x3ffd3ed00x400d8312:0x3ffd3f00 0x400d81a0:0x3ffd3f30

M5 Stack core 2
/lv_micropython/ports/esp32 make erase
/lv_micropython/ make  -C ports/esp32 LV_CFLAGS="-DLV_COLOR_DEPTH=16 -DLV_COLOR_16_SWAP=1" BOARD=M5CORE2 deploy
the build runs without any errors -> mount the image on esp - M5 Stack core 2
-> error on M5 Stack
Guru Meditation Error: Core 0' panic'ed (StoreProhibited). Exeption was unhandled.
Backtrace: 0x4019e1cb:0x3ffd345b00x400d88b2:0x3ffd45e0 0x400d872d:0x3ffd4610

I build the MicroPython Firmware out of the micropython repo and this worked when I mount the Firmware on the ESP
WROOM 32
/micropython/ports/esp32 make erase
/micropython/ make  -C ports/esp32  BOARD=GENERIC deploy
the build runs without any errors -> mount the image on esp - WROOM 32 
no problem microphython v 1.19.1 is shown

TTGO
/micropython/ports/esp32 make erase
/micropython/ make  -C ports/esp32  BOARD=LILYGO_TTGO_LORA32 deploy
the build runs without any errors -> mount the image on esp - TTGO
no problem microphython v 1.19.1 is shown

if you need more info, please ask!

amirgon, 2022-07-08:
> throw an error -> tool Xtensa -esp32-elf does not have a version compatible with platform linux-arm64

Are you building on arm host?
I only tried building on x86 hosts.

> Esp - WROOM32
> BOARD=GENERIC_SPIRAM deploy

As far as I know WROOM32 don't have SPI ram. Why are you building `GENERIC_SPIRAM` board?

> TTGO
> BOARD=GENERIC_SPIRAM deploy
> -> error on TTGO
> E(874) psram: PSRAM ID read error: 0xffffffff
> E(874) spiram: SPI RAM enabled but initialization failed. Bailing out.

Looks like your TTGO doesn't have SPI ram, so you can't use `GENERIC_SPIRAM` board.

> -> error on M5 Stack
> Guru Meditation Error: Core 0' panic'ed (StoreProhibited). Exeption was unhandled.
> Backtrace: 0x4019e1cb:0x3ffd345b00x400d88b2:0x3ffd45e0 0x400d872d:0x3ffd4610

Please parse the backtrace.  
To parse the backtrace you can pass the the hex values to xtensa-esp32-elf-addr2line:
```
xtensa-esp32-elf-addr2line -aipfC -e ./ports/esp32/<Board>/application.elf 0x... ... ..
```
Make sure you provide the correct ELF file under the right "<Board>" directory.

PGNetHun, 2022-07-20:
Hello @Damon3499 ,
Please make sure to upload the correct firmware on device! So, SPIRAM supported firmware to device which has really external PSRAM (M5Core2 for example), and simple non-SPIRAM firmware to other devices (TTGO T-Display for example).
See PSRAM for different ESP32 variants: https://en.wikipedia.org/wiki/ESP32#Surface-mount_module_boards

Basically, these 2 generic boards would fit for the 2 main cases (with and without PSRAM):
- GENERIC
- GENERIC_SPIRAM

I could build the M5CORE2 board successfully after a small fix ([PR 56](https://github.com/lvgl/lv_micropython/pull/56)), and also checked GENERIC and GENERIC_SPIRAM build processes, they work.

As you tried to build the MicroPython firmware in `linux-arm64` environment, and you received an error message, maybe you should check this page to install the correct tool:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-tools.html

If it helps, I can share my environment (and some help scripts), so you can have same, and maybe it will work or simplify some tasks:

- x86 (64 bit) Ubuntu 20 (running in Oracle VM VirtualBox)
- ESP-IDF 4.4.1 (tested with real M5Core2 device, it works with latest MicroPython 1.19)

Directory structure:
```
~
|- esp
|    |- esp-idf-4-3-3   (git: https://github.com/espressif/esp-idf/tree/v4.3.3)
|    |- esp-idf-4-4-1   (git: https://github.com/espressif/esp-idf/tree/v4.4.1)
|- src
|    |- build-lv_micropython
|    |- lv_micropython   (latest git master branch)
```

Help scripts: [build-lv_micropython.zip](https://github.com/lvgl/lv_binding_micropython/files/9146994/build-lv_micropython.zip)

Extract to `build-lv_micropython` folder, and switch to it: 
`cd ~/src/build-lv_micropython`

**Clean** build folders: `./clean.sh`

**Erase device** (set `BOARD` first in `env-variables.sh`): `./erase.sh`

**Build** GENERIC-SPIRAM board: `./build-generic-spiram.sh`

**Deploy** firmware to device (set `BOARD` first in `env-variables.sh`): `./deploy.sh`

**Monitor** device output (install before `picocom`): `./monitor.sh`

PS:
in `env-variables.sh` script many ESP-IDF versions can be found, because previously only ESP-IDF 4.2.2 worked for me with MicroPython 1.18, but now MicroPython 1.19 works with ESP-IDF 4.4.1 also. You can clone other ESP-IDF versions also, which will work for you, and update `env-variables.sh` script to use that, if it helps.

closed 2022-10-21 by amirgon