lvgl/lv_binding_micropython #223: Compile error when enable LVGL LV_USE_MSG in lv_conf.h

bl4kraven, 2022-07-15:
Hi,

### version
newest master branch 649ce43c23634a8cdf9d1db6ba8aa9800e6e6793

### error reproduce

enable LV_USE_MSG in lv_config.h

> /*1: Enable a published subscriber based messaging system */
#define LV_USE_MSG 1

### gcc error message

error file is lv_mpy.c

> build-sdl/lvgl/lv_mpy.c:33530:17: error: ‘mp_lv_msg_subsribe_obj’ redeclared as different kind of symbol
33530 | STATIC mp_obj_t mp_lv_msg_subsribe_obj(size_t mp_n_args, const mp_obj_t *mp_args, void *lv_func_ptr)
      |                 ^~~~~~~~~~~~~~~~~~~~~~
build-sdl/lvgl/lv_mpy.c:33522:46: note: previous definition of ‘mp_lv_msg_subsribe_obj’ was here
33522 | STATIC MP_DEFINE_CONST_LV_FUN_OBJ_STATIC_VAR(mp_lv_msg_subsribe_obj, 3, mp_lv_msg_subsribe, lv_msg_subsribe);

lv_mpy.c error parts:
```
STATIC mp_obj_t mp_lv_msg_subsribe(size_t mp_n_args, const mp_obj_t *mp_args, void *lv_func_ptr)
{
    void *user_data = mp_to_ptr(mp_args[2]);
    lv_msg_subscribe_cb_t cb = mp_to_ptr(mp_args[1]);
    uint32_t msg_id = (uint32_t)mp_obj_get_int(mp_args[0]);
    void * _res = ((void *(*)(uint32_t, lv_msg_subscribe_cb_t, void *))lv_func_ptr)(msg_id, cb, user_data);
    return ptr_to_mp((void*)_res);
}

STATIC MP_DEFINE_CONST_LV_FUN_OBJ_STATIC_VAR(mp_lv_msg_subsribe_obj, 3, mp_lv_msg_subsribe, lv_msg_subsribe);

/*
 * lvgl extension definition for:
 * void *lv_msg_subsribe_obj(uint32_t msg_id, lv_obj_t *obj, void *user_data)
 */

STATIC mp_obj_t mp_lv_msg_subsribe_obj(size_t mp_n_args, const mp_obj_t *mp_args, void *lv_func_ptr)
{
    void *user_data = mp_to_ptr(mp_args[2]);
    lv_obj_t *obj = mp_to_lv(mp_args[1]);
    uint32_t msg_id = (uint32_t)mp_obj_get_int(mp_args[0]);
    void * _res = ((void *(*)(uint32_t, lv_obj_t *, void *))lv_func_ptr)(msg_id, obj, user_data);
    return ptr_to_mp((void*)_res);
}
```
so it's a variable redefined error, mp_lv_msg_subsribe_obj  is redefined.

### why 

if enable LV_USE_MSG LVGL enable two native function : **lv_msg_subscribe and lv_msg_subsribe_obj**.

but gen_mpy.py create the micropython function obj lv_msg_subsribe_obj for lv_msg_subscribe , so confict to the native LVGL function lv_msg_subsribe_obj. 

so maybe gen_mpy.py must check micropython function obj name not confict with LVGL native function.

amirgon, 2022-07-15:
Thank you for reporting this!

Fixed, by renaming `_obj` to `_mpobj`.

closed 2022-07-15 by amirgon

amirgon, 2022-07-15:
See also: https://github.com/lvgl/lvgl/pull/3488

commit 65304b68 (2022-07-15): Rename _obj to _mpobj

Fixes #223
commit 950bdc14 (2022-07-15): Rename _obj to _mpobj

Fixes #223