lvgl/lv_binding_micropython #226: Runtime error with any example that uses event_loop on unix port

jath03, 2022-07-27:
I'm trying to get the examples working on a raspberry pi through the unix port, but keep running into this error whenever an example uses the event_loop function.  

```bash
Traceback (most recent call last):
  File "example3.py", line 37, in <module>
  File "display_driver.py", line 2, in <module>
  File "display_driver_utils.py", line 35, in __init__
  File "display_driver_utils.py", line 163, in init_gui
  File "display_driver_utils.py", line 42, in init_gui_SDL
  File "lv_utils.py", line 95, in __init__
  File "lv_timer.py", line 142, in init
  File "lv_timer.py", line 103, in timer_create
RuntimeError: timer_create_ error: -1 (errno = 22)
```

amirgon, 2022-07-27:
Hi @jath03 !

Please specify: on which version of lv_micropython it happens, how you built it, what display driver you are using, etc.   
Also provide information about the rpi board you are using, Linux version etc.

The error you are seeing means that [timer_create](https://man7.org/linux/man-pages/man2/timer_create.2.html) function returned the error `EINVAL`.  
According to the docs this means:

> Clock ID, sigev_notify, sigev_signo, or sigev_notify_thread_id is invalid.

In the code we call `timer_create` like this:

https://github.com/lvgl/lv_binding_micropython/blob/341134744b2a919d8a9e7773bba1bdec0fa21b44/driver/linux/lv_timer.py#L98-L101

So:
-  `clockid` = `CLOCK_MONOTONIC`
- `sigev_notify` = `SIGEV_SIGNAL`
- `sigev_signo` = `SIGRTMIN + sig_id`

Looks like Linux on your side has some issue with one of these arguments.

jath03, 2022-07-27:
lv_microython: commit 6ebf96a9032170a957dd07c83350480c08b79c78
built using the instructions in the README on the pi itself
SDL driver
Raspberry pi  3b
Linux 5.15.32-v7+

amirgon, 2023-06-10:
This is probably related to https://github.com/lvgl/lv_binding_micropython/issues/273, fixed in https://github.com/lvgl/lv_binding_micropython/pull/274.

Please reopen if it doesn't fix it.

closed 2023-06-10 by amirgon