lvgl/lv_binding_micropython #231: How to update the lvgl bar progressbar percentage when my other py function is running

bx5974, 2022-08-23:
I compiled lvgl for micropython 1.19.1

How to update the lvgl bar progressbar percentage when my other py function is running 

when i call my function - my function alone is running and lvgl progress bar is stuck 

Could you please help with an micropython lvgl progress bar updating example 

i was searching but I dont see any fitting micropython examples for this

bx5974, 2022-08-25:
@embeddedt can you point with an example?

amirgon, 2022-08-26:
> How to update the lvgl bar progressbar percentage when my other py function is running

The UI is updated by the [Event Loop](https://github.com/lvgl/lv_binding_micropython#the-event-loop).
By default, the event loop is running only when other Python code is **not running**. This allows us to run the GUI (and all other parts of the application) on a **single thread**.  

You should usually write event-driven code: never "wait in a loop forever" for example. If there is some task to do, a callback is called and returns as soon as possible in order to return the control to the event loop.

If you have some long-running sequence of command you want to run without blocking the GUI, your options are:
- Break your sequence to shorter event-driven sequences. uasyncio is one way to do it, there are other ways too.
- Run your long-running sequence in a [different thread](https://docs.micropython.org/en/latest/library/_thread.html).
- Create your own event loop and call it from your long-running sequence. This is usually more complicated and we don't recommend it.

closed 2022-10-21 by amirgon