lvgl/lv_binding_micropython #235: compille error in ports/unix, missing the `readline_hist`

stallion5632, 2022-09-14:
```
test@ubuntu:~/micropython/ports/unix$ make -j6
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
CC input.c
CC modtime.c
CC moduselect.c
CC alloc.c
CC fatfs_port.c
In file included from ../../py/mpstate.h:33:0,
                 from input.c:33:
input.c: In function ‘prompt_write_history’:
../../py/mpstate.h:293:40: error: ‘mp_state_vm_t {aka struct _mp_state_vm_t}’ has no member named ‘readline_hist’
 #define MP_STATE_VM(x) (mp_state_ctx.vm.x)
                                        ^
../../py/misc.h:127:34: note: in definition of macro ‘MP_ARRAY_SIZE’
 #define MP_ARRAY_SIZE(a) (sizeof(a) / sizeof((a)[0]))
                                  ^
./mpconfigport.h:54:23: note: in expansion of macro ‘MP_STATE_VM’
 #define MP_STATE_PORT MP_STATE_VM
                       ^~~~~~~~~~~
../../py/mpstate.h:293:40: error: ‘mp_state_vm_t {aka struct _mp_state_vm_t}’ has no member named ‘readline_hist’
 #define MP_STATE_VM(x) (mp_state_ctx.vm.x)
                                        ^
../../py/misc.h:127:47: note: in definition of macro ‘MP_ARRAY_SIZE’
 #define MP_ARRAY_SIZE(a) (sizeof(a) / sizeof((a)[0]))
                                               ^
./mpconfigport.h:54:23: note: in expansion of macro ‘MP_STATE_VM’
 #define MP_STATE_PORT MP_STATE_VM
                       ^~~~~~~~~~~
In file included from input.c:33:0:
../../py/mpstate.h:293:40: error: ‘mp_state_vm_t {aka struct _mp_state_vm_t}’ has no member named ‘readline_hist’
 #define MP_STATE_VM(x) (mp_state_ctx.vm.x)
                                        ^
./mpconfigport.h:54:23: note: in expansion of macro ‘MP_STATE_VM’
 #define MP_STATE_PORT MP_STATE_VM
                       ^~~~~~~~~~~
CC mpbthciport.c
../../py/mkrules.mk:80: recipe for target 'build-standard/input.o' failed
make: *** [build-standard/input.o] Error 1
make: *** Waiting for unfinished jobs....
```

then add the `const char *readline_hist[8];` into sturct of `../../py/mpstate.h:typedef struct _mp_state_vm_t`, it compiled success!

amirgon, 2022-09-15:
Hi @stallion5632  

Every commit [passes CI](https://github.com/lvgl/lv_binding_micropython/actions), so it definitely compiles and runs on Linux.

Are you trying to build your own Micropython with lv_binding_micropython?  
I suggest you have a look at [lv_micropython](https://github.com/lvgl/lv_micropython). We recommend using it directly, or at least as a reference if you are trying to build lv_binding_micropython with your custom micropython.

closed 2022-10-21 by amirgon

LinDong88, 2025-01-20:
try：const char *readline_hist[MICROPY_READLINE_HISTORY_SIZE];