lvgl/lv_binding_micropython #240: Compilation error on micropython v1.19.1

liangyingy, 2022-10-31:
I want to integrate lvgl on micropython, but it compiles wrongly, what should I do?

port: esp32
micropython: micropython/micropython v1.19.1

```
idf.py -D MICROPY_BOARD=GENERIC -B build-GENERIC  -DUSER_C_MODULES=../../../lib/lv_bindings/bindings.cmake build
Executing action: all (aliases: build)
Running cmake in directory /home/lewin/workspace/micropython/ports/esp32/build-GENERIC
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DMICROPY_BOARD=GENERIC -DUSER_C_MODULES=../../../lib/lv_bindings/bindings.cmake -DCCACHE_ENABLE=0 /home/lewin/workspace/micropython/ports/esp32"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/lewin/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/lewin/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/lewin/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/lewin/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/lewin/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/lewin/workspace/micropython/ports/esp32/build-GENERIC/sdkconfig
Loading defaults file /home/lewin/workspace/micropython/ports/esp32/build-GENERIC/sdkconfig.combined...
/home/lewin/workspace/micropython/ports/esp32/build-GENERIC/sdkconfig.combined:8 CONFIG_COMPILER_OPTIMIZATION_LEVEL_RELEASE was replaced with CONFIG_COMPILER_OPTIMIZATION_SIZE
/tmp/confgen_tmpw3qim5ps:69 line was updated to CONFIG_BTDM_CTRL_MODE_BR_EDR_ONLY=n
/tmp/confgen_tmpw3qim5ps:70 line was updated to CONFIG_BTDM_CTRL_MODE_BTDM=n
-- Found Python3: /home/lewin/.espressif/python_env/idf4.4_py3.8_env/bin/python3.8 (found version "3.8.10") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "micropython" version: v1.19-dirty
-- Adding linker script /home/lewin/workspace/micropython/ports/esp32/build-GENERIC/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/lewin/esp/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /home/lewin/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/lewin/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /home/lewin/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/lewin/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/lewin/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/lewin/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/lewin/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /home/lewin/esp/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
Including User C Module(s) from ../../../lib/lv_bindings/bindings.cmake
Found User C Module(s): usermod_lv_bindings, lvgl_interface
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test ieee802154 jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/lewin/esp/esp-idf/components/app_trace /home/lewin/esp/esp-idf/components/app_update /home/lewin/esp/esp-idf/components/asio /home/lewin/esp/esp-idf/components/bootloader /home/lewin/esp/esp-idf/components/bootloader_support /home/lewin/esp/esp-idf/components/bt /home/lewin/esp/esp-idf/components/cbor /home/lewin/esp/esp-idf/components/cmock /home/lewin/esp/esp-idf/components/coap /home/lewin/esp/esp-idf/components/console /home/lewin/esp/esp-idf/components/cxx /home/lewin/esp/esp-idf/components/driver /home/lewin/esp/esp-idf/components/efuse /home/lewin/esp/esp-idf/components/esp-tls /home/lewin/esp/esp-idf/components/esp32 /home/lewin/esp/esp-idf/components/esp_adc_cal /home/lewin/esp/esp-idf/components/esp_common /home/lewin/esp/esp-idf/components/esp_eth /home/lewin/esp/esp-idf/components/esp_event /home/lewin/esp/esp-idf/components/esp_gdbstub /home/lewin/esp/esp-idf/components/esp_hid /home/lewin/esp/esp-idf/components/esp_http_client /home/lewin/esp/esp-idf/components/esp_http_server /home/lewin/esp/esp-idf/components/esp_https_ota /home/lewin/esp/esp-idf/components/esp_https_server /home/lewin/esp/esp-idf/components/esp_hw_support /home/lewin/esp/esp-idf/components/esp_ipc /home/lewin/esp/esp-idf/components/esp_lcd /home/lewin/esp/esp-idf/components/esp_local_ctrl /home/lewin/esp/esp-idf/components/esp_netif /home/lewin/esp/esp-idf/components/esp_phy /home/lewin/esp/esp-idf/components/esp_pm /home/lewin/esp/esp-idf/components/esp_ringbuf /home/lewin/esp/esp-idf/components/esp_rom /home/lewin/esp/esp-idf/components/esp_serial_slave_link /home/lewin/esp/esp-idf/components/esp_system /home/lewin/esp/esp-idf/components/esp_timer /home/lewin/esp/esp-idf/components/esp_websocket_client /home/lewin/esp/esp-idf/components/esp_wifi /home/lewin/esp/esp-idf/components/espcoredump /home/lewin/esp/esp-idf/components/esptool_py /home/lewin/esp/esp-idf/components/expat /home/lewin/esp/esp-idf/components/fatfs /home/lewin/esp/esp-idf/components/freemodbus /home/lewin/esp/esp-idf/components/freertos /home/lewin/esp/esp-idf/components/hal /home/lewin/esp/esp-idf/components/heap /home/lewin/esp/esp-idf/components/idf_test /home/lewin/esp/esp-idf/components/ieee802154 /home/lewin/esp/esp-idf/components/jsmn /home/lewin/esp/esp-idf/components/json /home/lewin/esp/esp-idf/components/libsodium /home/lewin/esp/esp-idf/components/log /home/lewin/esp/esp-idf/components/lwip /home/lewin/workspace/micropython/ports/esp32/main /home/lewin/esp/esp-idf/components/mbedtls /home/lewin/esp/esp-idf/components/mdns /home/lewin/esp/esp-idf/components/mqtt /home/lewin/esp/esp-idf/components/newlib /home/lewin/esp/esp-idf/components/nghttp /home/lewin/esp/esp-idf/components/nvs_flash /home/lewin/esp/esp-idf/components/openssl /home/lewin/esp/esp-idf/components/openthread /home/lewin/esp/esp-idf/components/partition_table /home/lewin/esp/esp-idf/components/perfmon /home/lewin/esp/esp-idf/components/protobuf-c /home/lewin/esp/esp-idf/components/protocomm /home/lewin/esp/esp-idf/components/pthread /home/lewin/esp/esp-idf/components/sdmmc /home/lewin/esp/esp-idf/components/soc /home/lewin/esp/esp-idf/components/spi_flash /home/lewin/esp/esp-idf/components/spiffs /home/lewin/esp/esp-idf/components/tcp_transport /home/lewin/esp/esp-idf/components/tcpip_adapter /home/lewin/esp/esp-idf/components/tinyusb /home/lewin/esp/esp-idf/components/ulp /home/lewin/esp/esp-idf/components/unity /home/lewin/esp/esp-idf/components/usb /home/lewin/esp/esp-idf/components/vfs /home/lewin/esp/esp-idf/components/wear_levelling /home/lewin/esp/esp-idf/components/wifi_provisioning /home/lewin/esp/esp-idf/components/wpa_supplicant /home/lewin/esp/esp-idf/components/xtensa
-- Configuring done
CMake Error at /home/lewin/esp/esp-idf/tools/cmake/project.cmake:428 (add_executable):
  Cannot find source file:

    /home/lewin/workspace/micropython/ports/esp32/build-GENERIC/lv_mp.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx
Call Stack (most recent call first):
  CMakeLists.txt:58 (project)

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
cmake failed with exit code 1
make: *** [Makefile:34: all] Error 2
```

amirgon, 2022-10-31:
Why don't you fork and use [lv_micropython](https://github.com/lvgl/lv_micropython)?  
lv_micropython is Micropython fork where LVGL integration is already done.

liangyingy, 2022-10-31:
lv_micropython modifies micropython on the build system, which will make micropython update, and lv_micropython cannot adapt to the latest version of micropython very quickly.
And I think, lv_binding_micropython should be an independent module, it should work with micropython very quickly. Rather than being highly integrated with lv_micropython.

amirgon, 2022-10-31:
> And I think, lv_binding_micropython should be an independent module, it should work with micropython very quickly. Rather than being highly integrated with lv_micropython.

Unfortunately, today lv_binding_micropython is not a Micropython independent module since it requires some changes in Micropython itself to work. For example, it uses Micropython's GC for LVGL memory allocations.

If you don't want to use lv_micropython directly, you can diff lv_micropython vs. upstream Micropython, see the changes and adapt them on your Micropython fork.  

Here is a list of steps for integrating lv_binding_micropython in your Micropython fork (a little outdated): https://github.com/lvgl/lv_binding_micropython#adding-micropython-bindings-to-a-project

closed 2022-10-31 by liangyingy