lvgl/lv_binding_micropython #241: [NameError: name 'items' isn't defined] in c modules

TsXor, 2022-11-02:
error:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lib/myfont/myfont.py", line 61, in readchr
  File "/lib/myfont/myfont.py", line 31, in mapaddr
NameError: name 'items' isn't defined
```

that line:
```python
coffset = bisect_left(zhmapper, cord)
```

`bisect.c`
```c
#include "py/dynruntime.h"

STATIC mp_int_t bisect_left_impl(mp_obj_t a, mp_int_t x) {
    mp_int_t lo = 0; mp_int_t hi = mp_obj_get_int(mp_obj_len(a));
	mp_int_t mid; mp_obj_t curptr; 
    while (lo < hi) {
        mid = (lo + hi) >> 1; // 右移一位等价于地板除2
		curptr = mp_obj_subscr(a, mp_obj_new_int(mid), MP_OBJ_SENTINEL);
        if (mp_obj_get_int(curptr) < x) {
            lo = mid + 1;
        } else {
            hi = mid;
        }
    }
    return lo;
}

STATIC mp_obj_t bisect_left(mp_obj_t a_obj, mp_obj_t x_obj) {
    mp_obj_t a = a_obj;
    mp_int_t x = mp_obj_get_int(x_obj);

    mp_int_t result = bisect_left_impl(a, x);

    return mp_obj_new_int(result);
}

STATIC MP_DEFINE_CONST_FUN_OBJ_2(bisect_left_obj, bisect_left);

mp_obj_t mpy_init(mp_obj_fun_bc_t *self, size_t n_args, size_t n_kw, mp_obj_t *args) {
    MP_DYNRUNTIME_INIT_ENTRY

    mp_store_global(MP_QSTR_bisect_left, MP_OBJ_FROM_PTR(&bisect_left_obj));

    MP_DYNRUNTIME_INIT_EXIT
}

```

TsXor, 2022-11-02:
Just compiled like regular mpy, don't know how it is different in lvgl.

TsXor, 2022-11-02:
Magical! When I did a little change, the error changed:
```c
#include "py/dynruntime.h"

STATIC mp_int_t bisect_left_impl(mp_obj_t a, mp_int_t x) {
    mp_int_t lo = 0; mp_int_t mid;
	size_t hi; mp_obj_t *pyarr;
	mp_obj_get_array(a, &hi, &pyarr);
	mp_int_t cur; // here
    while (lo < hi) {
        mid = (lo + hi) >> 1; // 右移一位等价于地板除2
		cur = mp_obj_get_int(pyarr[mid]);
        if (cur < x) { // and here
            lo = mid + 1;
        } else {
            hi = mid;
        }
    }
    return lo;
}

STATIC mp_obj_t bisect_left(mp_obj_t a_obj, mp_obj_t x_obj) {
    mp_obj_t a = a_obj;
    mp_int_t x = mp_obj_get_int(x_obj);

    mp_int_t result = bisect_left_impl(a, x);

    return mp_obj_new_int(result);
}

STATIC MP_DEFINE_CONST_FUN_OBJ_2(bisect_left_obj, bisect_left);

mp_obj_t mpy_init(mp_obj_fun_bc_t *self, size_t n_args, size_t n_kw, mp_obj_t *args) {
    MP_DYNRUNTIME_INIT_ENTRY

    mp_store_global(MP_QSTR_bisect_left, MP_OBJ_FROM_PTR(&bisect_left_obj));

    MP_DYNRUNTIME_INIT_EXIT
}
```
error:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lib/myfont/myfont.py", line 61, in readchr
  File "/lib/myfont/myfont.py", line 31, in mapaddr
TypeError: expected tuple/list
```

amirgon, 2022-11-04:
Hi @TsXor !

It's a bit hard for me to say what is the problem by reading your code.  
I don't know what you are trying to do, or how you configured / changed / built the project etc.
If I could reproduce your problem on my side maybe I could help.

The please provide:
- [**Minimal, Reproducible Example**](https://stackoverflow.com/help/minimal-reproducible-example) for your problem. Please try to remove any line of code that is irrelevant to the problem.
- All the details for how you configure and build the project
- Step by step all the commands you run until your see this problem

TsXor, 2022-11-08:
> Hi @TsXor !
> 
> It's a bit hard for me to say what is the problem by reading your code. I don't know what you are trying to do, or how you configured / changed / built the project etc. If I could reproduce your problem on my side maybe I could help.
> 
> Please provide:
> 
>     * [**Minimal, Reproducible Example**](https://stackoverflow.com/help/minimal-reproducible-example) for your problem. Please try to remove any line of code that is irrelevant to the problem.
> 
>     * All the details for how you configure and build the project
> 
>     * Step by step all the commands you run until your see this problem

The code I posted is reproducible. You just have to compile it as a dynamic module with a proper `Makefile` like any of the ones in `examples/natmod/`
My `Makefile`:
```
# Location of top-level MicroPython directory
MPY_DIR = ../../../micropython

# Name of module
MOD = _unsquash_dat

# Source files (.c or .py)
SRC = _unsquash_dat.c

# Architecture to build for (x86, x64, armv7m, xtensa, xtensawin)
ARCH = xtensawin

# Include to get the rules for compiling and linking the module
include $(MPY_DIR)/py/dynruntime.mk
```

TsXor, 2022-11-08:
I can almost say that it is `mp_obj_subscr` that go wrong.