lvgl/lv_binding_micropython #251: couple of errors in gen_mpy.py code

kdschlosser, 2023-01-04:
function `get_arg_name` on line 2033 has an error on line 2037

```python
def get_arg_name(arg):
    if isinstance(arg, c_ast.PtrDecl) or isinstance(arg, c_ast.FuncDecl):
        return get_arg_name(arg.type)
    if hasattr(arg, 'declname'): return arg.declname
    if hasattr(arg, 'name'): return name
    return 'unnamed_arg'
```

I believe should read

```python
def get_arg_name(arg):
    if isinstance(arg, c_ast.PtrDecl) or isinstance(arg, c_ast.FuncDecl):
        return get_arg_name(arg.type)
    if hasattr(arg, 'declname'): return arg.declname
    if hasattr(arg, 'name'): return arg.name
    return 'unnamed_arg'
```

and function `gen_obj` on line 2501 has an error in the function on line 2520

```python
def gen_obj(obj_name):
    # eprint('Generating object %s...' % obj_name)
    is_obj = has_ctor(obj_name)
    should_add_base_methods = is_obj and obj_name != 'obj'
    obj_metadata[obj_name] = {'members' : collections.OrderedDict()}

    # Generate object methods
    for method in get_methods(obj_name):
        try:
            gen_mp_func(method, obj_name)
        except MissingConversionException as exp:
            gen_func_error(method, exp)

    # Generate object construction function, if needed
    if is_obj:
        ctor_func = get_ctor(obj_name)
        try:
            gen_mp_func(ctor_func, obj_name)
        except MissingConversionException as exp:
            gen_func_error(cctor_func, exp)

    # print([method.name for method in methods])
    ctor = """
STATIC mp_obj_t {obj}_make_new(
    const mp_obj_type_t *type,
    size_t n_args,
    size_t n_kw,
    const mp_obj_t *args)
{{
    return make_new(&mp_{ctor_name}_mpobj, type, n_args, n_kw, args);
}}
"""

    print("""
/*
 * {module_name} {obj} object definitions
 */
    """.format(
            module_name = module_name,
            obj = obj_name))

    print("""
STATIC const mp_rom_map_elem_t {obj}_locals_dict_table[] = {{
    {locals_dict_entries}
}};

STATIC MP_DEFINE_CONST_DICT({obj}_locals_dict, {obj}_locals_dict_table);

STATIC void {obj}_print(const mp_print_t *print,
    mp_obj_t self_in,
    mp_print_kind_t kind)
{{
    mp_printf(print, "{module_name} {obj}");
}}

{ctor}

STATIC const mp_lv_obj_type_t mp_lv_{obj}_type = {{
    {{
        {{ &mp_type_type }},
        .name = MP_QSTR_{obj},
        .print = {obj}_print,
        {make_new}
        {binary_op}
        .attr = call_parent_methods,
        .locals_dict = (mp_obj_dict_t*)&{obj}_locals_dict,
        {buffer_p}
        .parent = {parent}
    }},
#ifdef LV_OBJ_T
    {lv_class}
#endif
}};
    """.format(
            module_name = module_name,
            obj = sanitize(obj_name), base_obj = base_obj_name,
            base_class = '&mp_%s_type' % base_obj_name if should_add_base_methods else 'NULL',
            locals_dict_entries = ",\n    ".join(gen_obj_methods(obj_name)),
            ctor = ctor.format(obj = obj_name, ctor_name = ctor_func.name) if has_ctor(obj_name) else '',
            make_new = '.make_new = %s_make_new,' % obj_name if is_obj else '',
            binary_op = '.binary_op = mp_lv_obj_binary_op,' if is_obj else '',
            buffer_p = '.buffer_p = { .get_buffer = mp_lv_obj_get_buffer },' if is_obj else '',
            parent = '&mp_lv_%s_type.mp_obj_type' % parent_obj_names[obj_name] if obj_name in parent_obj_names and parent_obj_names[obj_name] else 'NULL',
            lv_class = '&lv_%s_class' % obj_name if is_obj else 'NULL',
            ))
```

should read 

```python
def gen_obj(obj_name):
    # eprint('Generating object %s...' % obj_name)
    is_obj = has_ctor(obj_name)
    should_add_base_methods = is_obj and obj_name != 'obj'
    obj_metadata[obj_name] = {'members' : collections.OrderedDict()}

    # Generate object methods
    for method in get_methods(obj_name):
        try:
            gen_mp_func(method, obj_name)
        except MissingConversionException as exp:
            gen_func_error(method, exp)

    # Generate object construction function, if needed
    if is_obj:
        ctor_func = get_ctor(obj_name)
        try:
            gen_mp_func(ctor_func, obj_name)
        except MissingConversionException as exp:
            gen_func_error(ctor_func, exp)

    # print([method.name for method in methods])
    ctor = """
STATIC mp_obj_t {obj}_make_new(
    const mp_obj_type_t *type,
    size_t n_args,
    size_t n_kw,
    const mp_obj_t *args)
{{
    return make_new(&mp_{ctor_name}_mpobj, type, n_args, n_kw, args);
}}
"""

    print("""
/*
 * {module_name} {obj} object definitions
 */
    """.format(
            module_name = module_name,
            obj = obj_name))

    print("""
STATIC const mp_rom_map_elem_t {obj}_locals_dict_table[] = {{
    {locals_dict_entries}
}};

STATIC MP_DEFINE_CONST_DICT({obj}_locals_dict, {obj}_locals_dict_table);

STATIC void {obj}_print(const mp_print_t *print,
    mp_obj_t self_in,
    mp_print_kind_t kind)
{{
    mp_printf(print, "{module_name} {obj}");
}}

{ctor}

STATIC const mp_lv_obj_type_t mp_lv_{obj}_type = {{
    {{
        {{ &mp_type_type }},
        .name = MP_QSTR_{obj},
        .print = {obj}_print,
        {make_new}
        {binary_op}
        .attr = call_parent_methods,
        .locals_dict = (mp_obj_dict_t*)&{obj}_locals_dict,
        {buffer_p}
        .parent = {parent}
    }},
#ifdef LV_OBJ_T
    {lv_class}
#endif
}};
    """.format(
            module_name = module_name,
            obj = sanitize(obj_name), base_obj = base_obj_name,
            base_class = '&mp_%s_type' % base_obj_name if should_add_base_methods else 'NULL',
            locals_dict_entries = ",\n    ".join(gen_obj_methods(obj_name)),
            ctor = ctor.format(obj = obj_name, ctor_name = ctor_func.name) if has_ctor(obj_name) else '',
            make_new = '.make_new = %s_make_new,' % obj_name if is_obj else '',
            binary_op = '.binary_op = mp_lv_obj_binary_op,' if is_obj else '',
            buffer_p = '.buffer_p = { .get_buffer = mp_lv_obj_get_buffer },' if is_obj else '',
            parent = '&mp_lv_%s_type.mp_obj_type' % parent_obj_names[obj_name] if obj_name in parent_obj_names and parent_obj_names[obj_name] else 'NULL',
            lv_class = '&lv_%s_class' % obj_name if is_obj else 'NULL',
            ))

```

dequi, 2023-01-05:
@kdschlosser OFF_TOPIC and desparate attempt: but since this is your most recent action here on github - Eventghost server and forum are dead and nobody is able to contact you. could you have a look.

amirgon, 2023-01-08:
Hi @kdschlosser, thank you for pointing out these problems.

Could you please open a PR instead of an issue for this?  
This would make it easier for us to review your suggested changes and test them.

If you can, it could also be useful to know what C code you processed with the script, that brought up each of these problems.

PGNetHun, 2023-10-11:
Hello!
- `cctor_func` typo is already fixed
- `name` typo will be fixed in PR: https://github.com/lvgl/lv_binding_micropython/pull/298

closed 2023-10-16 by kdschlosser