lvgl/lv_binding_micropython #252: LVGL *must* be compiled with LV_COLOR_DEPTH=16 (currently LV_COLOR_DEPTH=32

8fb790dd, 2023-01-05:
tried to build firmware for port RP2 with LV_COLOR_DEPTH=16  with repo lv_micropython, 
according documented command for ESP32:

make  -j -C ports/rp2 LV_CFLAGS="-DLV_COLOR_DEPTH=16"   BOARD=PICO USER_C_MODULES=../../lib/lv_bindings/bindings.cmake 

But maybe Makefile Parameters or Makefile for RP2 should be changed?
Do you have a hint for me, what i should change?
thx a lot in advance

Get the RuntimeError below by from __init__ of the generic python st77xx.py driver:
====
  File "libs/gen_st77xx.py", line 479, in __init__
  File "libs/gen_st77xx.py", line 448, in __init__
RuntimeError: LVGL *must* be compiled with LV_COLOR_DEPTH=16 (currently LV_COLOR_DEPTH=32.
MicroPython v1.19.1-657-g89c3b3943-dirty on 2023-01-05; Raspberry Pi Pico with RP2040
===

amirgon, 2023-01-08:
Hi @8fb790dd ,

The Makefile/CMake build system of the rp2 port probably doesn't handle `LV_CFLAGS` correctly.  
I'm not using rp2 so I'll wait for some rp2 user to fix this and send a PR.

For now, as a workaround, you can edit `lib/lv_bindings/lv_conf.h` and set `LV_COLOR_DEPTH` there to 16.

8fb790dd, 2023-01-13:
thx a lot for the workaround hint :-) 
as soon as i am a little bit more familiar with the lvgl lib and the lv_bindings machinery i will take a deeper look to the make/cmake ;-)

closed 2023-03-18 by amirgon