lvgl/lv_binding_micropython #259: lv_meter_t is not exposed to MicroPython

kdschlosser, 2023-02-19:
as the title states the lv_meter_t structure is not exposed to MicroPython. The rest of the structures for the meter are exposed.

amirgon, 2023-02-19:
No LVGL API function is receiving lv_meter_t, therefore it is not considered part of the API and not exposed by the bindings.   
Why do you need it?

kdschlosser, 2023-02-20:
Nothing I needed in the structure. I just went and looked at the code I noticed that everything in the structure is able to be changed by use of functions. I just noticed that it was not exposed like a lot of the other structures are and thought it worth mentioning. I understand why it is not exposed at this point. Sorry for opening an issue on this. I pulled the trigger on it before I dove into seeing what is going on in the code.

closed 2023-02-20 by kdschlosser