lvgl/lv_binding_micropython #262: AttributeError: 'textarea' object has no attribute 'add_event'

jd3096-mpy, 2023-03-09:
In docs ,there are many examples that can't run about textarea
`##### startup script #####

#!/opt/bin/lv_micropython -i

import lvgl as lv
import display_driver

##### main script #####

def ta_event_cb(e):
    code = e.get_code()
    ta = e.get_target_obj()
    if code == lv.EVENT.CLICKED or code == lv.EVENT.FOCUSED:
        # Focus on the clicked text area
        if kb != None:
            kb.set_textarea(ta)

    elif code == lv.EVENT.READY:
        print("Ready, current text: " + ta.get_text())

# Create the password box

pwd_ta = lv.textarea(lv.scr_act())
pwd_ta.set_text("")
pwd_ta.set_password_mode(True)
pwd_ta.set_one_line(True)
pwd_ta.set_width(lv.pct(45))
pwd_ta.set_pos(5, 20)
pwd_ta.add_event(ta_event_cb, lv.EVENT.ALL, None)

# Create a label and position it above the text box
pwd_label = lv.label(lv.scr_act())
pwd_label.set_text("Password:")
pwd_label.align_to(pwd_ta, lv.ALIGN.OUT_TOP_LEFT, 0, 0)

# Create the one-line mode text area
text_ta = lv.textarea(lv.scr_act())
text_ta.set_width(lv.pct(45))
text_ta.set_one_line(True)
text_ta.add_event(ta_event_cb, lv.EVENT.ALL, None)
text_ta.set_password_mode(False)

text_ta.align(lv.ALIGN.TOP_RIGHT, -5, 20)

# Create a label and position it above the text box
oneline_label = lv.label(lv.scr_act())
oneline_label.set_text("Text:")
oneline_label.align_to(text_ta, lv.ALIGN.OUT_TOP_LEFT, 0, 0)

# Create a keyboard
kb = lv.keyboard(lv.scr_act())
kb.set_size(lv.pct(100), lv.pct(50))

kb.set_textarea(pwd_ta)  # Focus it on one of the text areas to start

`
like so ,the same mistake

AttributeError: 'textarea' object has no attribute 'add_event'

jd3096-mpy, 2023-03-09:
It shoud be add_event_cb

amirgon, 2023-03-11:
Hi @jd3096-mpy,

Regarding the issue you raised about `add_event_cb`, I'd like to clarify that LVGL's master branch is the development branch and undergoes frequent changes. The lv_micropython master branch follows the LVGL master branch. If you're looking for a stable version, please use the [release/v8](https://github.com/lvgl/lv_micropython/tree/release/v8) branch. However, if you're interested in the latest and greatest version and can tolerate frequent API changes, you can use the master branch. We always strive to keep the master branch functional and healthy, but please note that it's not always aligned with lv_micropython master. That's why we use git-submodules. By updating git submodules recursively in lv_micropython, you can get the LVGL version that lv_micropython is aligned to, which is a recent version but not always the master branch.

In this case, the issue you're seeing is related to https://github.com/lvgl/lvgl/issues/4011. LVGL made changes to the API, one of which was renaming `add_event_cb` to `add_event`. When lv_micropython was aligned to an older LVGL version, this created a discrepancy between LVGL's master branch and documentation, and lv_micropython. Indeed, in that version of lv_micropython, you still need to use `add_event_cb`. However, I recently [merged these changes to lv_micropython's master branch](https://github.com/lvgl/lv_micropython/commit/cd1a2178d04b8f499a64492155bdd827991b37a5). Therefore, the examples you provided that use `add_event` should now work with the latest version of lv_micropython.

Also, please be aware that the online simulator has not yet been updated to reflect these changes. If you want to see the updated behavior, you can test it locally using any port other than the js port.

I hope this explanation helps to clear things up. If you have any further questions or concerns, please let me know.

closed 2023-03-11 by amirgon