lvgl/lv_binding_micropython #263: st77xx.py is broken

amirgon, 2023-03-17:
After adapting the Micropython Bindings to the new LVGL driver API (https://github.com/lvgl/lvgl/issues/4011), st77xx.py broke.

Related: https://forum.lvgl.io/t/issue-st77xx-py-generic-driver/11216

The problem is as following:

https://github.com/lvgl/lv_binding_micropython/blob/cf206e14253b2823522f139f3fa60b4219bd654f/driver/generic/st77xx.py#L438-L443

st77xx.py accesses `disp_drv.draw_buf.buf_act` directly. This is no longer possible because `draw_buf_act` is now a private LVGL member that is not exposed to the users. It is not part of the LVGL public API and therefore not part of the Micropython bindings.

The user is supposed to access the data through `color` argument which represents the buffer.

@eudoxos - What was the reason you used `buf_act` directly instead of `color`?

closed 2023-03-18 by amirgon

ngc6589, 2023-03-19:
Fix color.__dereference__ in st77xx.py

`self.blit(area.x1,area.y1,w:=(area.x2-area.x1+1),h:=(area.y2-area.y1+1),color.__dereference__(lv.color_t. __SIZE__*w*h),is_blocking=False)`

lv.color_t.__SIZE__ cannot be resolved.

Described at def __init__(....)
import lvgl as lv
import lv_utils
Better yet, move the import 2 line to the top of the file.

thank you.

eudoxos, 2023-03-19:
@amirgon IIRC `buf_act` is the current output buffer and `__dereference__` get its address (where the blitting starts). I dont't see how can this be replaced by `color`?

amirgon, 2023-03-19:
> lv.color_t.SIZE cannot be resolved.

@ngc6589 I changed it back to 2 instead of `lv.color_t. __SIZE__`.  
I think the original idea was to import and use LVGL only in `St77xx_lvgl` constructor, which makes the driver easily portable to other libraries and not LVGL specific, so I kept the imports in the constructor.

Please let me know if it works now, as I cannot test this on my side.

> I dont't see how can this be replaced by dereferencing `color`?

@eudoxos  Looking at the [C callback](https://github.com/lvgl/lvgl/blob/cdcd6cf5ef33ed1ad898c6d03f213fe4bb54ad49/src/core/lv_disp.h#L250-L251): 

```c
/**
 * Set the flush callback whcih will be called to copy the rendered image to the display.
 * @param disp      pointer to a display
 * @param flush_cb  the flush callback
 */
void lv_disp_set_flush_cb(lv_disp_t * disp, void (*flush_cb)(struct _lv_disp_t * disp, const lv_area_t * area,
                                                             lv_color_t * color_p));
```

`color_p` is a pointer to the display buffer, which is an array with element type of `lv_color_t`.   
I agree that this is not well documented, and the name "color" here is a poor choice...   (cc: @kisvegabor)
`draw_buf_act` is internal to LVGL and not part of the API. The driver is expected to access the `color_p` array instead.  

Here is an example from [Linux FB driver](https://github.com/lvgl/lvgl/blob/cdcd6cf5ef33ed1ad898c6d03f213fe4bb54ad49/src/dev/disp/fb/lv_linux_fbdev.c#L200):

```c
static void flush_cb(lv_disp_t * disp, const lv_area_t * area, lv_color_t * color_p)
{
...
lv_memcpy(&fbp32[location], (uint32_t *)color_p, (area->x2 - area->x1 + 1) * 4);
```

ngc6589, 2023-03-20:
> @ngc6589 I changed it back to 2 instead of lv.color_t. __SIZE__.
> I think the original idea was to import and use LVGL only in St77xx_lvgl constructor, which makes the driver easily portable to other [...]

I've seen a fix that changed lv.color_t.__SIZE__ back to 2.
As a result of confirmation, it worked without error.

kisvegabor, 2023-03-20:
> I agree that this is not well documented, and the name "color" here is a poor choice... (cc: @kisvegabor)
draw_buf_act is internal to LVGL and not part of the API. The driver is expected to access the color_p array instead.

What name would you find more intuitive? E.g. `rendered_img`?

amirgon, 2023-03-20:
> What name would you find more intuitive? E.g. `rendered_img`?

First, I think that documentation does not say anything about the callback parameters and their meanings.
Instead of `color`, even `buffer` is more intuitive for me, to stand for the buffer where the rendered image data is.

eudoxos, 2023-03-20:
Or perhaps `pixmap`, which conveys that it is a place for pixels?

kisvegabor, 2023-03-21:
`px_map`? We use the `px` abbreviation.

amirgon, 2023-03-21:
> `px_map`? We use the `px` abbreviation.

Sounds good to me. But please also add comments/docs that explain the callback arguments, so everything would be clear.

kisvegabor, 2023-03-22:
I updated it.

commit af1171e9 (2023-03-18): Fixes #263