lvgl/lv_binding_micropython #268: javascript build fails

kdschlosser, 2023-04-07:
This is the error I am getting

```
make: *** No rule to make target 'lib/lv_bindings/driver/SDL/sdl_common.c', needed by 'build/genhdr/qstr.i.last'.  Stop.
```

I am following the instructions outlined here

https://github.com/lvgl/lv_micropython/tree/lvgl_javascript_v9#javascript-port

embeddedt, 2023-04-07:
Known issue, I have not had time to update the JS port to work with v9 yet. (This error is fixed locally but GUIs can't be interacted with for another reason.)

kdschlosser, 2023-04-07:
how is there the working simulator running version 9 on the LVGL website?

embeddedt, 2023-04-07:
That version is 1.5 months old and doesn't include the refactors that broke the port.

kdschlosser, 2023-04-07:
Ahh  OK...

There needs to be a c compiler argument added to the Makefile

```
CFLAGS += -DNDEBUG -DMICROPY_SDL=1
```

That gets things going until lv_mpy.c compiles. then you will be hit with errors for memory_order and undefined variables. I am thinking this is in SDL2 but not sure. I used a version of gen_mpy that I adapted so names of objects to be ignored could be passed to the script so it wouldn't generate the code for those items. So that solved that issue. Now I am faced with linking errors which I have a feeling is the cause of your interaction problems.

```
build/lvgl/lv_mpy.c:8323:13: warning: function 'lv_disp_t_flush_cb_callback' has internal linkage but is not defined [-Wundefined-internal]
STATIC void lv_disp_t_flush_cb_callback(struct _lv_disp_t *disp_drv, const lv_area_t *area, lv_color_t *color_p);
            ^
build/lvgl/lv_mpy.c:10152:111: note: used here
            case MP_QSTR_flush_cb: dest[0] = mp_lv_funcptr(&mp_funcptr_flush_cb_mpobj, (void*)data->flush_cb, lv_disp_t_flush_cb_callback, MP_QSTR_lv_disp_t_flush_cb, data->user_data); break; // converting from callback void (*)(lv_disp_t *disp_drv, lv_area_t *area, lv_color_t *color_p);
                                                                                                              ^
build/lvgl/lv_mpy.c:9776:13: warning: function 'lv_disp_t_draw_ctx_init_callback' has internal linkage but is not defined [-Wundefined-internal]
STATIC void lv_disp_t_draw_ctx_init_callback(struct _lv_disp_t *disp, lv_draw_ctx_t *draw_ctx);
            ^
build/lvgl/lv_mpy.c:10166:126: note: used here
            case MP_QSTR_draw_ctx_init: dest[0] = mp_lv_funcptr(&mp_funcptr_draw_ctx_init_mpobj, (void*)data->draw_ctx_init, lv_disp_t_draw_ctx_init_callback, MP_QSTR_lv_disp_t_draw_ctx_init, data->user_data); break; // converting from callback void (*)(lv_disp_t *disp, lv_draw_ctx_t *draw_ctx);
                                                                                                                             ^
build/lvgl/lv_mpy.c:9777:13: warning: function 'lv_disp_t_draw_ctx_deinit_callback' has internal linkage but is not defined [-Wundefined-internal]
STATIC void lv_disp_t_draw_ctx_deinit_callback(struct _lv_disp_t *disp, lv_draw_ctx_t *draw_ctx);
            ^
build/lvgl/lv_mpy.c:10167:130: note: used here
            case MP_QSTR_draw_ctx_deinit: dest[0] = mp_lv_funcptr(&mp_funcptr_draw_ctx_init_mpobj, (void*)data->draw_ctx_deinit, lv_disp_t_draw_ctx_deinit_callback, MP_QSTR_lv_disp_t_draw_ctx_deinit, data->user_data); break; // converting from callback void (*)(lv_disp_t *disp, lv_draw_ctx_t *draw_ctx);
                                                                                                                                 ^
build/lvgl/lv_mpy.c:10108:13: warning: function 'lv_disp_t_wait_cb_callback' has internal linkage but is not defined [-Wundefined-internal]
STATIC void lv_disp_t_wait_cb_callback(struct _lv_disp_t *disp_drv);
            ^
build/lvgl/lv_mpy.c:10188:108: note: used here
            case MP_QSTR_wait_cb: dest[0] = mp_lv_funcptr(&mp_funcptr_wait_cb_mpobj, (void*)data->wait_cb, lv_disp_t_wait_cb_callback, MP_QSTR_lv_disp_t_wait_cb, data->user_data); break; // converting from callback void (*)(lv_disp_t *disp_drv);
                                                                                                           ^
4 warnings generated.
LINK build/firmware.js
wasm-ld: error: build/build/lvgl/lv_mpy.o: undefined symbol: lv_disp_t_flush_cb_callback
wasm-ld: error: build/build/lvgl/lv_mpy.o: undefined symbol: lv_disp_t_draw_ctx_init_callback
wasm-ld: error: build/build/lvgl/lv_mpy.o: undefined symbol: lv_disp_t_draw_ctx_deinit_callback
wasm-ld: error: build/build/lvgl/lv_mpy.o: undefined symbol: lv_disp_t_wait_cb_callback
wasm-ld: error: build/build/lvgl/lv_mpy.o: undefined symbol: lv_disp_t_flush_cb_callback
wasm-ld: error: build/build/lvgl/lv_mpy.o: undefined symbol: lv_disp_t_draw_ctx_init_callback
wasm-ld: error: build/build/lvgl/lv_mpy.o: undefined symbol: lv_disp_t_draw_ctx_deinit_callback
wasm-ld: error: build/build/lvgl/lv_mpy.o: undefined symbol: lv_disp_t_wait_cb_callback
```

embeddedt, 2023-04-07:
I have fixed all of the compilation errors locally; I just haven't pushed the changes as otherwise GitHub will compile and release a broken build to the website. The issue is that the ticking logic got rewritten for the SDL driver and so `lv_task_handler` isn't being called regularly anymore like it should.

kdschlosser, 2023-04-07:
Oh ok...

embeddedt, 2023-04-07:
Hoping to fix it today as I have some free time now. :slightly_smiling_face:

kdschlosser, 2023-04-07:
That would be cool. I have been screwing around with this thing for the last couple of weeks trying to get it to compile. I thought I was almost at the end but I guess not. I would have ended up running into the issue with the task handler.

kdschlosser, 2023-04-07:
I want to be able to run the simulator with more memory. I keep on hitting the 1mb memory cap with the simulator on the website.

kdschlosser, 2023-04-07:
I have tried using the unix port and that's a nogo with WSL.  It keeps on bombing on me with a segmentation fault error.

kdschlosser, 2023-04-07:
I was toying around with the idea of writing in the ability to connect the JS binding with a circuit simulator. something like this

https://lushprojects.com/circuitjs/circuitjs.html

The circuit simulator would have some MCUs built into it. This would give a full tilt simulation where inputs and outputs could be simulated.

embeddedt, 2023-04-07:
As of the latest commits the  `lvgl_javascript_v9` branch should build and run properly again. :tada:

closed 2023-04-07 by embeddedt

kdschlosser, 2023-04-07:
yay!!!

amirgon, 2023-04-13:
> As of the latest commits the `lvgl_javascript_v9` branch should build and run properly again. 🎉

Cool! 👍