lvgl/lv_binding_micropython #269: Mac OS port does not work anymore

daviel, 2023-04-12:
I was trying to test the newer master branch and encountered following issue when running the example in the readme.

import lvgl as lv
lv.init()
from lv_utils import event_loop
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "lv_utils.py", line 45, in <module>
RuntimeError: Missing machine.Timer implementation!

It looks like it is using the [fallback](https://github.com/lvgl/lv_binding_micropython/blob/master/driver/linux/lv_timer.py) in the lv_timer if machine.Timer is not available. This leads to binding the timer from libc.so/librt.so via FFI which is not available under Mac OS. 

In version 8 of lvgl I could get lv_micropython to work without problems.

amirgon, 2023-04-13:
Hi @daviel, thank you for reporting this bug.

The event loop is required for version 9 of lvgl on all platforms, whether embedded or not. Since Micropython Timer and libc.so/librt.so timer functions are missing for MacOS, it seems that we'll need to implement a MacOS-specific Timer class in the same manner as `lv_timer.py` implements it for Linux and Linux-like systems.

Would you like to give implementing this a try? If you or anyone else would like to attempt it, I'd be happy to review a pull request.

daviel, 2023-04-16:
I will take a look at it but can't promise to have a solution anytime soon. :)