lvgl/lv_binding_micropython #270: > did you use the repo I linked to or did you use the files I attached in a zip?? If you used the zipped files don't, use the linked repo instead

a0908896067, 2023-04-16:
> did you use the repo I linked to or did you use the files I attached in a zip?? If you used the zipped files don't, use the linked repo instead

Sorry，I git the wrong repo branch.
Now it can be compiled without errors,I will test it later.

_Originally posted by @jd3096-mpy in https://github.com/lvgl/lv_binding_micropython/issues/227#issuecomment-1356641632_

a0908896067, 2023-04-16:
I'm stuck here (still has many compilation errors) and I'm glad to see someone is OK, and I appreciate if someone can teach me how to do it without any compliation errors for the esp32-s3.  I'm good at assembly , C, Arduino programming but just get started how to use Micropython & LVGL. I 'm not familiar with (or I don't know) the binding between lv_micropython and the link[  ](https://github.com/kdschlosser/lv_binding_micropython/tree/esp32-s-c-h_support) from kdschlosser. Per kdschlosser, can't use .zip file , then how to combine both of lv_micropython & lv_binding_micropython in one directory without .zip file. Appreciate if someone can give me steps by steps like, thanks in advance

git clone https://github.com/lvgl/lv_micropython.git
cd lv_micropython
git submodule update --init --recursive lib/lv_bindings (this step will download data from lv_micropython, not from [     ](https://github.com/kdschlosser/lv_binding_micropython/tree/esp32-s-c-h_support) how to change ? )

Next you should build mpy-cross

make -C mpy-cross

Michael

amirgon, 2023-06-10:
For questions and help please use the forum: https://forum.lvgl.io/c/micropython/8

closed 2023-06-10 by amirgon