lvgl/lv_binding_micropython #271: garbage collection in the simulator not working

kdschlosser, 2023-04-20:
The garbage collection in the online simulator is not functioning

https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/02b56115173e8824cd0ac97947c32cf190a01e92/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/02b56115173e8824cd0ac97947c32cf190a01e92/examples/widgets/label/lv_example_label_4.py&script_direct=3ea10711be6512c21cfa1b9155efd3cec5ccc7d3

you can se the first 2 prints of the amount of free memory but not the 3rd. It gets stuck when garbage collection is happening. Also automatic garbage collection doesn't occur either.

embeddedt, 2023-04-20:
The garbage collector on the JS port has never worked 100% reliably, unfortunately. I tested your example on MicroPython's current version of JS port (without the LVGL additions) and there `gc.collect()` crashes immediately ([this has also been reported upstream](https://github.com/micropython/micropython/issues/10692#issuecomment-1422252187)). Supposedly the garbage collector was fixed a while ago but it seems to have been broken again.

kdschlosser, 2023-04-21:
ok cool. Just thought it was worth mentioning

closed 2023-06-10 by amirgon

embeddedt, 2023-06-22:
It looks like https://github.com/micropython/micropython/pull/11823 would fix this, I'll be keeping an eye on it and may cherry-pick it to the JS branch if upstream won't merge it.